heitzmann/gdspy #4: Python v3.5

johnlb, 2016-05-05:
FYI, I've started developing with this on Python 3.5 and I haven't had any trouble so far.

Had to get VS14 for pip to install (I'm on Win10), but it was fine after that. Probably could generate a wheel for it though, if you wanted?

heitzmann, 2016-05-05:
Sounds great @johnlb! If you could send the wheel I'll make it available with the current release.

johnlb, 2016-05-05:
Looks like you're using distutils, so I think I can't make a wheel? Instead, what's your current build process? I can just repeat that.

Also, looks like master is ahead of the most recent release (0.7.1). Would you rather I use an older commit inline with 0.7.1 or just build from the head?

heitzmann, 2016-05-05:
I use the following the compile an installation file for both 32 and 64 versions with Microsoft SDK 7.1 (python version 2.7 needs the 7.0):
```
set DISTUTILS_USE_SDK=1
setenv /x64 /release
C:\Python\python.exe setup.py build --plat-name=win-amd64 bdist --format=wininst
setenv /x86 /release
C:\Python_32\python.exe setup.py build --plat-name=win32 bdist --format=wininst
```

I'm on linux, so I'll probably investigate this in more detail only for the next release.

As for the version, it would be better to compile the 0.7.1 version, so that all installers behave the same.

johnlb, 2016-05-07:
Sorry for the delay, I've actually been having some trouble with the build (if its not one thing...). Should get it worked out soon.

johnlb, 2016-05-07:
Here's the 0.7.1 release. I'm not able to compile the 0.8.1 release (either myself or through pip). It is getting caught on the clipper library:

building 'gdspy.clipper' extension
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\BIN\cl.exe /c /nologo /Ox /W3 /GL /DNDEBUG /MT -IC:\Users\John\AppData\Local\Programs\Python\Python35-32\include -IC:\Users\John\AppData\Local\Programs\Python\Python35-32\include /EHsc /Tpgdspy/clipper.cpp /Fobuild\temp.win-amd64-3.5\Release\gdspy/clipper.obj
clipper.cpp
gdspy/clipper.cpp(4750): error C2065: 'M_PI': undeclared identifier
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio 14.0\\VC\\BIN\\cl.exe' failed with exit status 2

[gdspy-0.7.1-py3.5.zip](https://github.com/heitzmann/gdspy/files/253637/gdspy-0.7.1-py3.5.zip)

johnlb, 2016-05-07:
Turns out, the fix was simple. Had to add "#define _USE_MATH_DEFINES" to clipper.hpp. Will send a pull request in a sec.

Here is 0.8.1

[gdspy-0.8.1.win32-py3.5.zip](https://github.com/heitzmann/gdspy/files/253642/gdspy-0.8.1.win32-py3.5.zip)

heitzmann, 2016-05-09:
Thanks @johnlb! I'll merge your chengaes as well.

closed 2016-05-09 by heitzmann

aloter, 2016-05-24:
Thank you both for providing the wheel, but I for some reason it didn't work for me.
I have Python 3.5 64bit (Windows), after wheel install I get:

> ImportError: cannot import name 'boolext'

As above the issue was

> gdspy/clipper.cpp(4750): error C2065: 'M_PI': undeclared identifier

I tried the fix suggested above, and it worked for me. When building I got a few warnings, but gdspy was installed.
Not sure why the above files didn't work. Let me know if you'd like to debug.

johnlb, 2016-05-27:
Hey -

Sorry for the delay. I'm too busy this week, but I'll take a look at generating a new binary early next week.