heitzmann/gdspy #9: Wrong results from fast_boolean()

aeberspaecher, 2016-08-23:
Using production layouts, we have observed wrong results using fast_boolean() using the "and" operation. The error seems to be related to fracturing in such way that if the clipper results are to be fractured, the results may be wrong.

 Is it possible that clipper results violate some condition implicit to gdspy?

Unfortunately, I cannot provide a test case by now. If I can spare some time in the coming weeks, I'll try to find a simple example useful for hunting down the bug.

heitzmann, 2016-08-23:
Hello @aeberspaecher, it is possible indeed, but without a test case it is unlikely I can figure it out... I'll wait for your input in the coming weeks.

aeberspaecher, 2016-11-11:
Unfortunately I won't be able to deliver a test case any time soon as I'm out of office for an extended time span. I think it's best to close this issue until I'm back and can get help in preparing an example that does not include any of our production data.

closed 2016-11-11 by aeberspaecher