heitzmann/gdspy #11: Using DEPLOF font

amccaugh, 2016-08-26:
I'd like to add the [DEPLOF](http://mems.technion.ac.il/) font to gdspy.

[image]

It's a font used on other layout editors (e.g. [LayoutEditor](http://www.layouteditor.net/wiki/Credits) [CleWin](http://www.wieweb.com/nojava/downloadframe.html) ), and is purpose-built for 2D layout/fab.  I received permission from the creator of the MATLAB gdsii-toolbox (Ulf Griesmann) to adapt his implementation, and have it basically ready to go.  My question is whether how I should write the commit--I was thinking of adding an additional argument to the __init__ argument of the Text class.  So that instead of 

`def __init__(self, text, size, position=(0, 0), horizontal=True, angle=0, layer=0, datatype=0)`

it would now be

`def __init__(self, text, size, position=(0, 0), horizontal=True, angle=0, **font = 'block',** layer=0, datatype=0)`

where `font` could be chosen to be either 'block' (the current font) or 'deplof' (the proposed font), so the change would be transparent and no one's layouts would be changed, but users would now have the option to select the DEPLOF font.  (and perhaps down the road the default could be changed to DEPLOF)

Does sound reasonable?

heitzmann, 2016-08-26:
I could not find the license for this font. Is it compatible to GPL3?

amccaugh, 2016-08-26:
There doesn't seem to be any license in particular aside from the language
"free for non-profit use", so I emailed asking for direct permission from
the author, I'll post when he gets back to me.

amccaugh, 2016-08-29:
I received explicit permission to use the font from the author in the gdspy GPLv3 code.  Beyond that, I'm not sure how to evaluate whether it's compatible with the GPLv3.  I would figure permission is the most important factor

heitzmann, 2016-08-29:
In that case, I think it's ok to submit the PR.
The signature you proposed looks good, but instead of `'block'` it might be better to use `None` for the default. In the future the function is could be extended to use any font installed in the system and in that case `'block'` could conflict with an actual `'block'` font.

amccaugh, 2016-08-29:
Sounds good.  I still have to work out some sizing issues to match the existing code, but I'll be submitting a patch in the next few days

closed 2016-08-29 by heitzmann