heitzmann/gdspy #13: Unable to install gdspy v1.0 on Windows

applepie12, 2016-09-21:
Having some issues installing gdspy v1.0 on Windows, my version is Python 3.5.2 :: Anaconda 4.0.0 (64-bit).
Tried pip install, and got a " Failed building wheel for gdspy" error.
Tried the to pre-compiled binaries install option using gdspy-1.0.win-amd64-py3.5.exe, and getting "Python version 3.5 -32 required, which was not found in the registry" error. Made sure I did execute the 64-bit version.

rosivda, 2016-09-22:
We found the installation in Windows 10 (64-bit) to be tricky but were ultimately able to get it to work. We used IntelliJ IDEA with a Python plugin. Instead of Anaconda, we use WinPython on Windows. Please see instructions below, if you'd like to give it a try.

1. Download [WinPython 3.5.2.2](https://sourceforge.net/projects/winpython/files/WinPython_3.5/3.5.2.2/WinPython-64bit-3.5.2.2.exe/download).
2. Download [iPython 4.2.1](https://github.com/ipython/ipython/archive/4.2.1.zip).
3. Download [Gdspy 1.0](https://github.com/heitzmann/gdspy/releases/download/v1.0/gdspy-1.0.win-amd64-py3.5.exe).
4. Open ```WinPython Control Panel.exe``` and uninstall the latest ipython package.
5. In the control panel, install iPython 4.2.1 and Gdspy 1.0.
6. Append the path to ```python.exe``` (for example, C:\Users\USERNAME\WinPython-64bit-3.5.2.2\python-3.5.2.amd64) to the PATH Windows environmental variable.
7. Restart IntelliJ IDEA (tested with version 2016.2.4)
8. Verify installation by invoking ```import gdspy``` in the IDEA python console.

applepie12, 2016-09-23:
Thank you! Works for me. After installation can copy the gdspy and  gdspy-1.0-py3.5.egg-info to the  Anaconda3 lib folders and works as well.

okianus, 2016-09-23:
@heitzmann, I was able to install Gdspy 1.0 on Windows 10 (64-bit) with WinPython 3.5.2.2 and IntelliJ IDEA, following @rosivda's steps. However, gdspy behaves distinctly different than other python packages. 

This is evidenced by the requirement to append the path to ```python.exe``` to PATH environment variable. Without it, gdspy results in ImportError (in contrast to other packages, which are OK). WinPython distribution is portable, and hence relies on an isolated and self-consistent environment, without external dependencies. Is there anything in the build process of gdspy that could be missing, preventing it from being self-consistent?

heitzmann, 2016-09-23:
I really can't say... I just followed the instructions from setuptools and the python documentation. Maybe I can incorporate a more automatic buid process such as the one used in https://github.com/python-pillow/Pillow, but I won't have time to implement or test that for a while, sorry.

If anyone feels like  helping out on this, you are welcome!

heitzmann, 2017-01-20:
I've just uploaded a new version of gdspy and the respective windows installers.
If anyone has chance to check these, please let me know if they work. They did on my Windows install.
Cheers!

rosivda, 2017-01-21:
@heitzmann I would like to provide you with an early feedback with regard to gdspy 1.1. I have installed ```gdspy-1.1.win-amd64-py3.5.exe``` using ```WinPython-64bit-3.5.2.2``` on Windows 10 and have run initial checks against our existing libraries. 
The installation part is flawless - everything went smoothly. I appreciate the fact that the code base is now in Python 3.
I have noted that there must have been some functional changes to the ```gdspy.Cell``` class that have prevented our libraries from executing fully. I am seeing exceptions in ```gdspy.Cell.cell_dict```, ```cell.copy```, etc. that we will likely need to modify to comply with the new release.

heitzmann, 2017-01-21:
Hello @rosivda, sorry, I had to change that so that GdsLibrary could be used without further changes in "usual" scripts... yours are probably more complex than most people's! In any case, you should be able to fix most problems by changing `Cell.cell_dict` to `current_library.cell_dict`.

Thanks for testing the installer as well!

closed 2017-01-27 by heitzmann