heitzmann/gdspy #15: Problem with 'boolext'

Mathieu4, 2016-10-25:
Hi!

I'm a new user of gdspy and I have some issues with the latest version. I'm using Anaconda with Python 3.5 and every time I try to run my script I have this error prompting : 

>   File "C:\Users\Mathieu\Documents\Python\gdspy-1.0\gdspy\__init__.py", line 44, in <module> from gdspy import boolext

> ImportError: cannot import name 'boolext'

And I have no idea why. Do you ?

heitzmann, 2016-10-25:
Hello! It seems your module hasn't been properly installed. Did you use the widows pre-compiled binaries?

If you are trying to install from source you'll have to compile the c extensions yourself (boolext & clipper).

Mathieu4, 2016-10-25:
Thank you for your answer! Actually, I downloaded the source code (zip) file called gdspy-&.0.zip (because I can't run .exe on my computer). 

And how could I compile it ? I really don't know how to do it unfortunately :(

heitzmann, 2016-10-25:
Unfortunately if you can't run an exe in you computer, I'm afraid you won't be able to compile it.
You'd need to install the Visual Studio tools appropriate to your python version (see, for example, https://blog.ionelmc.ro/2014/12/21/compiling-python-extensions-on-windows/).

The easiest way by far is to use the precompiled binaries...

Mathieu4, 2016-10-25:
Thank you for help. I already have Visual Studio installed on my computer but I don't know how to use it. 

Anyway, I will find another way to create gds!

closed 2016-10-25 by Mathieu4