heitzmann/gdspy #16: Saving only 1 cell + references to file

Engi4, 2016-11-16:
Hay there

I currently have 3 GDSII files that I am combining into one layout. I have managed to get all 3 cells into one cell and transposed and rotated them as necessary. Now I would only like to save one the one cell I have to a new GDS file. 

import gdspy

file1 = gdspy.GdsImport('Dcsfq.gds')
file2 = gdspy.GdsImport('Conf2.gds')
file3 = gdspy.GdsImport('Tff4b.gds')

cell1 = file1.extract(file1.top_level()[0])
cell2 = file2.extract(file2.top_level()[0])
cell3 = file3.extract(file3.top_level()[0])

ref_cell = gdspy.Cell('TRANSLATED')
ref_cell.add(gdspy.CellReference(cell2, (150,0)))
ref_cell.add(gdspy.CellReference(cell1, (0,0)))
ref_cell.add(gdspy.CellReference(cell3, (225,-150), rotation=90))

newgds = gdspy.GdsPrint('new_gds.gds', unit=1.0e-6, precision=1.0e-9)
newgds.write_cell(ref_call)
newgds.close()

This however gives me an empty GDSII file. If someone could assist me with this I would be very greatful (I am new to gdspy and GDSII files in general)

Thanks

okianus, 2016-11-16:
@Engi4, I haven't used ```gdspy.GdsPrint()```, but have implemented the same functionality with ```gdspy.gds_print()```. Replace the last three lines of your code with:
```
gdspy.gds_print('new_gds.gds', cells=[ref_cell], unit=1.0e-6, precision=1.0e-9)
```

Engi4, 2016-11-16:
I believe I found my issue. The GDS files I am reading has more than one layer. So I need to find a way to loop through each layer transpose each one and then add them to the cell. So the issue is a bit bigger than i though.

heitzmann, 2016-11-17:
@Engi4, I believe both your code and @okianus's should work. Maybe you should try to check the contents of your cells with [LayoutViewer](http://gdspy.readthedocs.io/en/latest/io.html#layoutviewer).

Engi4, 2016-11-25:
That was exactly what I needed. Thank you very much @okianus!

closed 2016-11-25 by Engi4