heitzmann/gdspy #18: Adding all elements to an array

Engi4, 2016-11-25:
Good day

I want to add every element in a GDS file to an array. Currently I am doing this:

file1 = gdspy.GdsImport('Dcsfq.gds')
cell = file1.extract(file1.top_level()[0])

//Found this at another issue topic
references = []
for element in cell.elements:
    if isinstance(element, gdspy.CellReference):
        print 'Cell ' + element.ref_cell.name + ' referenced at ' + str(element.origin)
        references.append((element.ref_cell, element.origin))
    elif isinstance(element, gdspy.CellArray):
       print 'Array of cells ' + element.ref_cell.name + ' referenced at ' + str(element.origin)
       references.append((element.ref_cell, element.origin))

This gives me the list of all the element references as it should, but I need to know the locations of the polys and other basic elements as well. I think the problem lies with the fact that I am only getting the top layer. Is there a way to loop through the file to obtain the location and type (polys, slices etc) and put it all on an array.

For instance if I basically want to replicate an entire gds file (I know I use simpler methods, this is just for explaining purposes) and i want to find the location of each element in the file and make a new cell containing all the elements and their original locations. 

The reason I want to do this, is because I have multiple files that I want to merge, but they contain elements with the same names so i cant just use

ref_cell.add(gdspy.CellReference(cell2, (150,0)))
ref_cell.add(gdspy.CellReference(cell1,(0,0)))
ref_cell.add(gdspy.CellReference(cell3, (225,-150), rotation=90))

as then I get double naming. So if there is an easier way then I am proposing, I am open to try anything.

Sorry for the long post and thanks for any feedback.

heitzmann, 2016-11-25:
Hello @Engi4, if your goal is only to merge different files that may have similar cell names, the simplest solution is to use the `rename` argument in [gds_import](http://gdspy.readthedocs.io/en/latest/io.html#gdsimport).

Something around these lines (I modified the example you presented in #16):

```python
import gdspy

operations = {
    'Dcsfq.gds': {'origin': (150, 0)},
    'Conf2.gds': {'origin': (0, 0)},
    'Tff4b.gds': {'origin': (225, -150), 'rotation': 90}
}

ref_cell = gdspy.Cell('TRANSLATED')

for i, file_name in enumerate(operations):
    # import file to get cell names
    gdsii = gdspy.GdsImport(file_name)
    # append all cell names with the 'i' counter
    names = {c: '{}_{}'.format(c, i) for c in gdsii.cell_dict}
    # just out of curiosity:
    print('\nRenaming cells in ' + file_name + ':')
    print(names)

    # import again renaming all cells and extract the result
    gdsii = gdspy.GdsImport(file_name, rename=names)
    cell = gdsii.extract(gdsii.top_level()[0])

    # apply the respective operation
    ref_cell.add(gdspy.CellReference(cell, **operations[file_name]))

gdspy.gds_print('new_gds.gds', cells=[ref_cell], unit=1.0e-6, precision=1.0e-9)
```

Engi4, 2016-11-25:
Thank you so much! You are a scholar and gentleman!

I learned so much from your replies and from your library in general. Thank you very much.

closed 2017-01-20 by heitzmann