heitzmann/gdspy #21: Gdspy 1.1 installation on windows

aloter, 2017-01-23:
Following up on release 1.1 on Windows 7, python 3.5 64bit:

1. Installer (gdspy-1.1.win-amd64-py3.5.exe) doesn't work for me, error: Python 3.5-32 is required for this package.

2. When using pip I get the following error:
line 26, in <module> with open('README.md') as fin: FileNotFoundError: [Errno 2] No such file or directory: 'README.md'

Using Setuptools v34 and ez-setup 0.9.

Note that Installing gdspy1.0 works flawlessly with pip on this setup.

Thanks for the new version.

amccaugh, 2017-01-25:
I get the same error on Mac OS X and Python 2.7 (Anaconda clean install)

rosivda, 2017-01-25:
@aloter Have you tried installing [gdspy-1.1.win-amd64-py3.5.exe](https://github.com/heitzmann/gdspy/releases/download/v1.1/gdspy-1.1.win-amd64-py3.5.exe) using [WinPython 3.5.2.2](https://sourceforge.net/projects/winpython/files/WinPython_3.5/3.5.2.2/WinPython-64bit-3.5.2.2.exe/download) ? For me, the install was flawless (Windows 10 Version 1607).

aloter, 2017-01-27:
@rosivda thanks for the tip.
When trying to install these preconfigured packages such as Anaconda, other tools have experience some issues. I'm using a pure Python 3.5 and installed all the packages using pip or wheels.

Thanks!

heitzmann, 2017-01-27:
Thanks all for the feedback. The issue with the missing README file should be fixed in v1.1.1 now.
I can't figure out the problem with the Windows installer... All 6 versions work fine for me in their respective versions of WinPython.

aloter, 2017-01-27:
It's working flawlessly, thanks Lucas!

`>>> print('Using gdspy module version ' + gdspy.__version__)
Using gdspy module version 1.1.1
>>>`

closed 2017-01-27 by aloter