heitzmann/gdspy #22: Gdspy 1.1 write_gds() for hierarchical files

okianus, 2017-03-01:
@heitzmann I have installed the latest release (Patch 1.1.1) on Mac OSX, and am observing a different behaviour for the GDSII writing functionality compared to previous releases.

In the past (release 1.0), `gds_print()` would accept a list of cells to be included in the output file, and then append any cell dependencies to the list of cells to be written. Here's the relevant code:
<pre><code>if cells == None:
        cells = iter(Cell.cell_dict.values())
    else:
        cells = [Cell.cell_dict.get(c, c) for c in cells]
        i = 0
        <b>while i < len(cells):
            for cell in cells[i].get_dependencies():
                if cell not in cells:
                    cells.append(cell)
            i += 1</b>
        ...
        for cell in cells:
             outfile.write(cell.to_gds(unit / precision))</code></pre>
This worked flawlessly for deeply hierarchical cell structures. 

The implementation of this functionality in 1.1.1 (`write_gds()` function), however, does not explicitly add the dependencies:
<pre><code>if cells is None:
        cells = self.cell_dict.values()
    else:
        cells = [self.cell_dict.get(c, c) for c in cells]
        for cell in cells:
             outfile.write(cell.to_gds(unit / precision))</code></pre>
resulting in an empty GDSII file. If I add the while / for block (the part in bold above), the dependencies are indeed added, and the resulting file contains all intended hierarchy. 

Should the dependencies be explicitly added here, as before? Or am I missing something in the new implementation of `GdsLibrary`?

heitzmann, 2017-03-02:
@okianus Indeed this behavior is correct: only explicitly given cells are written (see the notes in http://gdspy.readthedocs.io/en/latest/library.html#gdspy.GdsLibrary.write_gds).

This adds flexibility for when the library must be split in several files (to be honest I've never used this GDSII feature myself, but it is sometimes needed and defined in the format specs).

okianus, 2017-03-02:
@heitzmann Thanks for the quick reply!

I can overcome this behavior by explicitly getting all the referenced cells using `get_dependencies()` and thus achieve the needed behaviour. But out of curiosity --  in what cases one would NOT want the dependencies to be outputted to the GDSII file? If I have a `top_cell` that references a few subcells, `write_gds(outfile, [top_cell])` would produce an incomplete structure. If the library is split into several files, can dependencies exist across more than one file? If that's the case, I'd appreciate a link to the specs. If not, then it appears like there is no reason for the dependencies to be excluded.

heitzmann, 2017-03-02:
Because GDSII is used in some very old equipment, file sizes sometimes matter. In order to keep sizes constrained it is possible to reference other GDSII files (similar to cell referencing).

In that case I could place my main cell in file 1 and referenced cells in files 2, 3... etc., so that they have smaller sizes. As I said, I have never used this feature myself, although I have faced the file size issue before when using an old laser writer.

The exact spec can be found here: http://boolean.klaasholwerda.nl/interface/bnf/gdsformat.html#rec_reflibs

okianus, 2017-03-03:
@heitzmann Thank you for the clarification and the link to the specs.

closed 2017-03-04 by okianus