heitzmann/gdspy #23: Boolean subtraction errors

okianus, 2017-03-10:
@heitzmann I am observing errors in the boolean subtraction algorithm in the latest gdspy 1.1.1. 

Errors seem to occur with any structures that contain holes. An example below creates two polygon sets, in a shape of the digits 8 and 6. It then subtracts two identical cells. gdspy 1.0 produces an empty result, as expected. On the other hand, the boolean `not` algorithm in gdspy 1.1.1 generates incorrect triangular structures.

```
import gdspy

coords8 = [[[-1.16635, -0.5    ], [-1.16635,  0.5    ], [-0.99945,  0.5    ], [-0.99945, -0.5    ]],
           [[-0.99945,  0.3331 ], [-0.99945,  0.5    ], [-0.50005,  0.5    ], [-0.50005, -0.5    ],
            [-0.99945, -0.5    ], [-0.99945, -0.3331 ], [-0.66695, -0.3331 ], [-0.66695, -0.08345],
            [-0.99945, -0.08345], [-0.99945,  0.08345], [-0.66695,  0.08345], [-0.66695,  0.3331 ]]]
coords6 = [[[ 0.50005, -0.5    ], [ 0.50005,  0.5    ], [ 1.16635,  0.5    ], [ 1.16635,  0.3331 ], [ 0.66695,  0.3331 ], [ 0.66695, -0.5    ]],
           [[ 0.66695, -0.5    ], [ 1.16635, -0.5    ], [ 1.16635,  0.08345], [ 0.66695,  0.08345], [ 0.66695, -0.08345], [ 0.99945, -0.08345], [ 0.99945, -0.3331 ], [ 0.66695, -0.3331 ]]]
poly8_layer1 = gdspy.PolygonSet(coords8, 1)
poly6_layer1 = gdspy.PolygonSet(coords6, 1)
poly8_layer2 = gdspy.PolygonSet(coords8, 2)
poly6_layer2 = gdspy.PolygonSet(coords6, 2)

cell1 = gdspy.Cell('CELL1')
cell2 = gdspy.Cell('CELL2')
cell1.add(poly8_layer1)
cell1.add(poly6_layer1)
cell2.add(poly8_layer2)
cell2.add(poly6_layer2)
cell_ref1 = gdspy.CellReference(cell1)
cell_ref2 = gdspy.CellReference(cell2)

result = gdspy.fast_boolean(cell_ref1, cell_ref2, 'not')
result_cell = gdspy.Cell('DIFF_CELL')
result_cell.add(result)
gdspy.write_gds('boolean_subtract_test_example.gds', cells=[cell1, cell2, result_cell])   # for gdspy 1.1.1
gdspy.print_gds('boolean_subtract_test_example.gds', cells=[result_cell])               # for gdspy 1.0
```

rosivda, 2017-03-10:
@heitzmann Unfortunately the above issue is holding up our migration to 1.1.1 (from 1.0). Was there some other reason to change the boolean code? We have not experienced any errors in gdspy 1.0.

heitzmann, 2017-03-19:
Hello @okianus and @rosivda,

The problem seems to have been the last update on the external library we use, which introduced the bug on their side. I've updated gdspy again to include the newest version, and the example you provided is working correctly for me.

I'm releasing a bugfix version 1.1.2. Please, let me know if the problem persists and thanks for the report!

rosivda, 2017-03-19:
@heitzmann Thanks for your quick fix. We will do the installation and verification of 1.1.2 shortly and will let you know how it goes. Since we rely on gdspy for our production builds, we have almost finished creating an automated testing framework that validates the integrity of all our binary artifacts as we transition from one release of gdspy to the next. This is related to issue #24 and is based on SHA-1 hashing of binary .gds files. Because of the sheer size of our libraries, we are likely to detect a wide range of bugs and inconsistencies between the releases (though not able to pass judgment about the significance of observed differences).

rosivda, 2017-03-31:
@heitzmann After testing many geometries, we have concluded that, unfortunately, the offset algorithm still has issues (1.1.2 tested, the issue may have existed previously). It is not a frequent error and does not prevent us from effectively using gdspy but it does occur. The picture below shows the original structures in cyan and the resultant offset operation in black (small offset was applied) - obviously incorrect.  Unfortunately, these errors are embedded into large layouts.  We are currently working on isolating the issue to a simplest presentable case that can be acted on. I am also testing a few theories about what circumstances trigger this and we will be reporting on this as we tackle this issue.

[image]

okianus, 2017-03-31:
@heitzmann Below is an example that illustrates the `offset` problem described by @rosivda above. It creates a few waveguide structures, and enlarges them by a small amount. Please note that the exact manifestation of the errors differs, depending on how much of the input structure is included in the example, but you can see an example of the type of errors that we observe in the output file. Thank you for your help with this.

<details><summary>Example of offset error:</summary><p>

```
import gdspy
import numpy as np

# structure coordinates
poly_stuctures = [
    np.array([[-7770.333,  3098.327],
       [-7771.369,  3100.531],
       [-7772.402,  3102.737],
       [-7773.431,  3104.944],
       [-7774.456,  3107.153],
       [-7775.478,  3109.364],
       [-7776.495,  3111.576],
       [-7777.509,  3113.791],
       [-7778.518,  3116.007],
       [-7779.524,  3118.225],
       [-7780.526,  3120.444],
       [-7781.525,  3122.666],
       [-7782.519,  3124.889],
       [-7783.509,  3127.113],
       [-7784.496,  3129.34 ],
       [-7785.479,  3131.568],
       [-7786.458,  3133.798],
       [-7787.433,  3136.03 ],
       [-7788.404,  3138.263],
       [-7789.371,  3140.498],
       [-7790.334,  3142.735],
       [-7791.294,  3144.973],
       [-7792.25 ,  3147.213],
       [-7793.201,  3149.455],
       [-7794.149,  3151.698],
       [-7795.093,  3153.943],
       [-7796.033,  3156.19 ],
       [-7796.969,  3158.438],
       [-7797.901,  3160.688],
       [-7798.829,  3162.939],
       [-7799.754,  3165.192],
       [-7800.674,  3167.447],
       [-7801.591,  3169.703],
       [-7802.503,  3171.961],
       [-7803.412,  3174.221],
       [-7804.317,  3176.482],
       [-7805.218,  3178.744],
       [-7806.114,  3181.008],
       [-7807.007,  3183.274],
       [-7807.896,  3185.541],
       [-7808.781,  3187.81 ],
       [-7809.663,  3190.08 ],
       [-7810.54 ,  3192.352],
       [-7811.413,  3194.626],
       [-7812.282,  3196.901],
       [-7813.148,  3199.177],
       [-7814.009,  3201.455],
       [-7814.866,  3203.734],
       [-7815.72 ,  3206.015],
       [-7816.569,  3208.298],
       [-7817.415,  3210.581],
       [-7818.256,  3212.867],
       [-7819.094,  3215.153],
       [-7819.927,  3217.442],
       [-7820.757,  3219.731],
       [-7821.583,  3222.022],
       [-7822.404,  3224.315],
       [-7823.222,  3226.609],
       [-7824.036,  3228.904],
       [-7824.845,  3231.201],
[... 2405 lines cut ...]
       [-9514.846,  3275.252],
       [-9512.175,  3274.987],
       [-9509.504,  3274.718]]),
np.array([[ -9664.941,   3279.339],
       [-10613.225,   3279.339],
       [-10613.225,   3282.739],
       [ -9664.941,   3282.739]]),
np.array([[-10613.225,   3279.339],
       [-10692.121,   3279.339],
       [-10692.121,   3282.739],
       [-10613.225,   3282.739]])]

# construct the cell structure
cell = gdspy.Cell('CELL')
cell.add(gdspy.PolygonSet(poly_stuctures))

# offset the structure
poly_offset = gdspy.offset(gdspy.CellReference(cell), distance=0.35, layer=1, join_first=True, precision=0.0001)
cell_offset = gdspy.Cell('CELL_OFFSET')
cell_offset.add(poly_offset)

# write the result to gds file
gdspy.write_gds('offset_test_output.gds', cells=[cell, cell_offset])
```
</p></details>

explorer2011, 2017-04-02:
Just a small comment - applying various geometrical operations to arbitrary complex layouts is a challenging task. Doing simple things on simple layouts (like sizing a rectangle) is simple, but doing this on geometries containing very small angles, or acute angles, or holes, etc. is non-trivial. The best software for this sort of things, in my opinion, is Calibre (from Mentor Graphics - now part of Siemens).

I tried to do resizing (you call it offsetting) of your example in another tool (klayout) - it avoided the problems that gdspy had (apparently, algorithms bugs), but had some other issues (narrow inward triangular cuts caused by two neighboring edges that are almost parallel).

In one of my previous companies, we had out own computational geometry library, and we had to continuously improve it until all issues were resolved (and new issues were showing up for some new unusual geometries in GDS files).

One unrelated note - you have some (I guess - unintended) jogs at the polygon fracture edges...

rosivda, 2017-04-03:
@explorer2011 With regard to building generic algorithms to handle arbitrary geometries – I agree, no mean feat! Gdspy handles the absolute majority of cases correctly, but in some corner cases (roughly, <1 per 100,000 structures), residual issues need to be worked out. We all come from backgrounds where we used various layout tools – and certainly have an option of applying them here as well. From our experience, however, we have not been happier to have built our integrated design framework on top of python/gdspy and have cut out all the other layout dependencies from our workflow. This allowed us to fully automate our design and fabrication cycle and collaborate more effectively in spite of running on different OSs and hardware.
The “jogs” that you may have observed are intended. Our waveguide rendering tool compensates for the displacement of the mode in a curved waveguide relative to the waveguide center. In cases when the curvature of the waveguide changes abruptly, the renderer will produce a kink in the waveguide to align the modes in the waveguides with different curvatures.

explorer2011, 2017-04-03:
Agreed, simplicity and ease of use is the king...

heitzmann, 2017-04-03:
It seems that the problem is in fact in the `fracture` code, not `offset`.
If I use `max_points=1e300` the problem stops. I have to check the code and see if I find the bug... I don't know how hard that will be, so it could take a while. @okianus, thanks for the example!

rosivda, 2017-04-03:
@heitzmann I have an observation - please take it with a grain of salt. I observe errors occurring only in the regions where a densely sampled structure ( such as a curved waveguide) intersects a very sparsely sampled structure (such as a very elongated rectangle only defined by its corners). @okianus 's example falls into that category.

aeberspaecher, 2017-04-15:
I have seen errors with fracturing for issue #9 as well. Unfortunately I still cannot provide test data.

rosivda, 2017-05-09:
@heitzmann I would like to add another piece of info. In order for us to overcome the above errors, we introduce a minute inclination in what should have been perfectly horizontal waveguides. This breaks the symmetry and allows gdspy to execute without errors.

heitzmann, 2017-08-11:
Sorry, I've been quite busy and this is a complex issue probably with the `_chop` algorithm for heavily degenerate polygons.

The simplest solution I can think of is to use the boolean C extension library to robustly do the fracturing of the polygons. This adds another `precision` setting for all functions depending on fractures, but I guess it's a fair price for having the correct result.

The changes are included in commit 693dfd8 (branch next). I'll leave the issue open until this fix is properly tested.

In the future I might add a `_chop` function in C to speed up a little compared to the current version.

closed 2017-10-21 by heitzmann

Miakatt, 2019-10-10:
Hi. Was this issue properly resolved and tested?  
I've been having problems with GDS designed developed in PICWriter which uses gdspy. Some designs crash the e-beam during writing and I suspect it has something to do with loops in the grating couplers.

heitzmann, 2019-10-10:
Hi @Miakatt, indeed this has been resolved. The _chop function is performed by the library clipper (C extension, external dependency), in which we haven't found any bugs so far. Do you have any more information on the issue? Or maybe an example gds file where the crash occurs (as minimal as possible, so we can investigate)?

Miakatt, 2019-10-10:
Hi @heitzmann.  I am using a Raith Voyager E-beam. My designs are created in PICWriter (https://github.com/DerekK88/PICwriter) in which I create a cladding layer (1), a top layer (2) and a mask layer (layer 3). Layer 3 is what I write with the E-Beam. 
This is done with the PICwriter Toolkit call to build_mask. This function does the boolean operation in gdspy:
 
```
  if wgt.resist=='+':
        cell.add(gdspy.fast_boolean(pWG, pCLAD, 'xor', precision=0.001, max_points=199, layer=fl, datatype=fd))
    elif wgt.resist=='-':
        cell.add(gdspy.fast_boolean(pWG, pCLAD, 'and', precision=0.001, max_points=199, layer=fl, datatype=fd))
```

I run the pattern through the Raith consistency check. It shows some errors, but these are usually extremely small features/corners that the E-beam can't write. Our expert says that these errors shouldn't be a problem in practice. 

I've included a link GDS of one of the patterns that locked up. It failed on the bottom-most grating coupler. I can't say at exactly which position, but it looked to be at the GC itself, not the waveguide. 

https://www.dropbox.com/s/thnkmcm7sil4mhs/GC_Sweep_straight_endfire_lock-up.gds?dl=0

I converted the pattern to ASCII in the Raith software and tried again, but it stopped writing at the same point. 

The GDS has 11 grating couplers of varying periodicity. I could provide a single GC, but I know this particular file failed twice. To determine if a more minimalist example also fails will require a lot of work as it only fails during writing in the ebeam.  

We're talking to a Raith engineer today, so hopefully they can look in to this too.

Miakatt, 2019-10-10:
For completeness, I've provided a png of the grating coupler area I think was the point of failure. 
I wonder if the issue is due to the joint in the GC, and shortening the grating region may prevent the lock-up. 
I shall let you know if we get any useful information from Raith. 

[image]

heitzmann, 2019-10-10:
The shapes look fine to me in KLayout... could it be that the problem is in the 4nm strip at the last grating tooth? Or maybe the shape is just too complex for the Raith software... in that case, maybe chop the taper part out, than chop the teeth from the outer cladding to avoid creating holes?

rosivda, 2019-10-10:
@Miakatt FYI, I have looked at your coupler layout using our custom .gds viewer. I am not able to observe a problem or what could be causing it. As you may be aware, during ebeam writing, gds file typically gets flattened by the writing software to a base structure that is a trapezoid. We have encountered cases when the flattened structure exploded to several Gb causing the ebeam buffer overflowing. The type of structure that you are showing, unless a part of a much more complex layout, should be ok.

Miakatt, 2019-10-10:
@heitzmann, @rosivda . Thank you for taking the time to look in to this. I am also able to view the layout in various third party software (KLayout, Layout Editor, MEEP etc) as well as the Raith Voyager viewer/editor. The issue does seem to be related to how the Voyager writing software interprets the .gds file. I wasn't aware of the 'explosive' memory requirements during writing. This is interesting, as the task manager shows a flat 5GB RAM usage, even though there is 32GB available. So, maybe it's a local issue?

I'm told that other users of the E-beam are doing more complex layouts (designed in Matlab) with no issues. I don't know the details of these files as I'm an commercial user of the University's facilities. They may be complex, but very modular/repetitive, and therefore not so complex for the Voyager software to compose. 

I will need to look in to this and also see what Raith come back with. For now, it looks like the issue is either with them, or a more local problem. Thank you again for checking the example gds. I will post back when I get more information.