heitzmann/gdspy #26: Validity of bounding box after cell deep copy

okianus, 2017-04-27:
`gdspy.Cell.copy()` provides the ability to create a deep copy of a cell. It was noticed that when this is done for nested cells, any `CellReferences` within the cell have a their `_bb_valid` flags set to `True`. Hence, if you perform a deep copy of a cell, and then try to get its bounding box using `get_bounding_box()`, the algorithm fails, generating a `KeyError` at the return statement of `gdspy.Cell.get_bounding_box()` method.

I do not know all the implications of this, but at the very least, I believe the `_bb_valid` flags for the dependencies have to be explicitly set to `False` within `cell.copy()` to allow recalculation of the bounding boxes: 
```
for ref in new_cell.get_dependencies(True):
            if ref._bb_valid:
                print('Reference ' + str(ref) + ' bb_valid flag is True, set to False')
                ref._bb_valid = False
```

heitzmann, 2017-05-05:
Hi @okianus,

I believe you r solution is correct, but I have to take a look into any side-effects that could be involved.

I'm quite dissatisfied with the current state of the bounding box logic because of those side-effects. For instance, if the bounding box is calculated and then a change is made in the internal array of points of any polygon, the bounding box of any cells involved will not reflect it... I guess that's the price we pay for caching the values.

heitzmann, 2017-05-06:
Fixed in 2dd455777df4f786ee9994716f84b96879f045ae
Thank you for the bugfix!

closed 2017-05-06 by heitzmann