heitzmann/gdspy #27: Open a gds, access manipulate existing elements(polygons etc) and cells.

vijaysi2chip, 2017-05-05:
Hi, 
            I am new to python and gdspy, I am writing a code to edit existing gds file, manipulate and dump new gds file, please help/guide me how can i achieve that. 

thanks
vijay

heitzmann, 2017-05-05:
Hello @vijaysi2chip,

Gdspy is not designed to edit GDSII files, so it is not as user-friendly in this respect as it is for creating new files.

That said, when you import an existing layout all shapes will be objects of type `Polygon` or `PolygonSet`. Rotations and translations of those objects can be easily done using the existing respective methods, but anything more complex will probably require modifying their internal data structure (`Polygon.points` and `PolygonSet.polygons`). In that case I suggest reading the source code to understand what you are doing exactly.

vijaysi2chip, 2017-05-05:
Thanks a lot for quick response, I will try to work with this, 

thanks again!

closed 2017-05-05 by vijaysi2chip