heitzmann/gdspy #30: Problem in copy and relocate the object after rotation.

vishalvashistha87, 2017-08-04:
[image]
[image]

I am trying to make rectangular shape array structure (all rectangular shape are same size), whose rotation angle is decided by formula. I am directly calculating the rotate value using a formula in radian. When the rotation angle is small, everything is ok, rectangular shaped fixed at particular location, but if theta values is large, it is making mistake to fixed the location of rectangular shape, more precisely the coordinate of each rectangular location changes with large theta values. I do not understand the reason, I am quite new in Python. Attaching both files for reference.  I am not sure that it's my programming mistake or related to this package. 
I will be happy if you could answer. 

`
# coding: utf-8

# In[1]:

import numpy
import gdspy
import math
print('Using gdspy module version ' + gdspy.__version__)

## ------------------------------------------------------------------ ##
##      POLYGONS
## ------------------------------------------------------------------ ##

## First we need a cell to add the polygons to.
poly_cell = gdspy.Cell('POLYGONS')

# Polygon points for blue. (220 by 22 nm with periodicity = 250 nm).
# points = [(0.015, 0.136), (0.114, 0.136), (0.114, 0.235), (0.136, 0.235), (0.136, 0.136), (0.235, 0.136), (0.235, 0.114),(0.136, 0.114), (0.136, 0.015), (0.114, 0.015), (0.114, 0.114), (0.015, 0.114)]

## Create the polygon on layer 1.

trans_cell = gdspy.Cell('TRANS')
# oper_cell = gdspy.Cell('OPERATIONS')
# poly1 = gdspy.Polygon(points, 2) 				# add first polygon to layer 1
# poly1 = gdspy.Polygon(points, 2).rotate(numpy.pi) 		# Rotate by 45 degree (numpy.pi/4)

nm=1e-9;
um=1e-6;

#rect_1 = gdspy.Rectangle( (0.0, 0.0), (250*nm, 95*nm), 1 ).rotate(3.14/6)	# Define horigontal rectangle

ncells = 5
P = 325*nm
Lambda  = 532*nm
c = 3e8
f = 50*um
#A = 10

for iind in xrange (-ncells, ncells):
    for jind in xrange(-ncells, ncells):
	#print( jind) 
	rel_x1 = iind*P				     # New location of the next polygons.
        rel_y1 = jind*P
	print(rel_x1,rel_y1)
	temp = (rel_x1**2) + (rel_y1**2)
	print(temp)	
	temp1 = ((rel_x1**2) + (rel_y1**2) + (f**2))**(0.5)
	print(temp1)
	Theta = 100*numpy.pi*(f-temp1)/Lambda     #    ****When 100 times, the location of rect changes.**** 
	print(Theta)
	rect_1 = gdspy.Rectangle( (0.0, 0.0), (250*nm, 95*nm), 1 ).rotate(Theta)
        poly1 = gdspy.copy(rect_1, rel_x1,rel_y1)    # Copy the first polygons into new location rel_x and rel_y.
        trans_cell.add(poly1)			     # Adding them in layer. If you dont add them nothing will appear.
	

## Same goes for Labels & Text

#text1 = gdspy.Text('BLUE 220 X 22 ' , 6, (15, -10), horizontal=True, angle=0, layer=1)       # Horizontal is false meaning vertical text. 

#trans_cell.add(text1)

# Addisional label can be created from here.
#label1 = gdspy.Label('BLUE 220 X 22 ',(15, -15), 'nw', layer=1)
#trans_cell.add(label1)

## ------------------------------------------------------------------ ##
##      OUTPUT
## ------------------------------------------------------------------ ##

## Output the layout to a GDSII file (default to all created cells).
## Set the units we used to micrometers and the precision to nanometers.
gdspy.write_gds('lens.gds', unit=1.0, precision=1.0e-12)

## ------------------------------------------------------------------ ##
##      IMPORT
## ------------------------------------------------------------------ ##

## ------------------------------------------------------------------ ##
##      VIEWER
## ------------------------------------------------------------------ ##

## View the layout using a GUI.  Full description of the controls can
## be found in the online help at http://gdspy.sourceforge.net/
gdspy.LayoutViewer()

# In[ ]:
`

heitzmann, 2017-08-04:
@vishalvashistha87 The problem here is that the rotation is applied around the origin (0, 0), and your rectangles are not centered in (0, 0). A quick solution is to use:
```python
rect_1 = gdspy.Rectangle((-125*nm, -47.5*nm), (125*nm, 47.5*nm), 1).rotate(Theta)
```
Another suggestion I'd offer is that you are duplicating all rectangles when using `copy`, which can become an issue if your design increases: you'll soon run out of memory.

How about:

```python
rect_1 = gdspy.Rectangle((-125*nm, -47.5*nm), (125*nm, 47.5*nm), 1).rotate(Theta).translate(rel_x1, rel_y1)
trans_cell.add(rect_1)
```

closed 2017-08-04 by heitzmann

vishalvashistha87, 2017-08-06:
@heitzmann , Million thanks dear. Our group use your tool quite frequently. I would like to asked one more question. Does it possible to find the boundary of an arbitrary shape (some jpeg image) by this tool and than fill it with some specific size disk depending upon the color of that location ? This design tool can be very useful for photonics community.

heitzmann, 2017-08-06:
@vishalvashistha87 I'm glad to hear that!

To work with images you'll need other libraries, GDSPY specializes in GDSII only. There are several options out there, depending on your exact problem and preference:

- http://scikit-image.org/
- https://docs.scipy.org/doc/scipy/reference/tutorial/ndimage.html
- http://docs.opencv.org/3.0-beta/
- https://python-pillow.org/
- http://luispedro.org/software/mahotas/