heitzmann/gdspy #31: Problem importing in Anaconda on Ubuntu, python 3.6

FaustinCarter, 2017-10-19:
I've tried:

`pip install gdspy`

and also

`git clone github:gdspy.git`

followed by

`pip install -e gdspy`

The install goes fine, but the importing fails every time with the following message:

`gdspy/clipper.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_createERmm`

This works fine under Anaconda on OS X with python 3.6.

heitzmann, 2017-10-19:
@FaustinCarter have you tried using the system python?
I don't use Anaconda, so I'm not sure I can help you there without more information.
Can you provide the installation log? Maybe we can find the problem there.

FaustinCarter, 2017-10-19:
OK, after doing some stumbling around and figuring out how to get pip to be verbose and then reading the log and doing some more googling, I found this: http://discourse.mc-stan.org/t/pystan-anaconda-installation-instructions/1557 where folks were having a similar problem. The short version is that Anaconda's libgcc is for an older version of gcc than the one that ships with a new Ubuntu install (maybe Apple is still using an older gcc, which is why it works there?). Anyhow:

I updated libgcc in Anaconda `conda update libgcc` and now everything is fine! I'm changing this issue to a request to add a statement to the installation instructions that people using Anaconda should try updating their libgcc if they get import errors!

heitzmann, 2017-10-19:
Glad to hear you've figured it out!

closed 2017-10-19 by heitzmann