heitzmann/gdspy #33: LayoutViewer hangs when I try to close it.

mdaal, 2017-10-30:
Im using Conda ipython  with python version  3.6.3 on MacOS 10.13 to run code which opens a LayoutViewer.  The code  successfully opens a viewer. However when I try to close the viewer, either by clicking the red button at the upper left of the window or by selecting python>"quit python", I get a spinning beachball. The ipython command prompt returns (is no longer blocked) despite the spinning beachball.

I have the same problem in my python 2.7 conda environment. 

Additionally, if I ignore the spinning beachball and hanging LayoutViewer, and continue working in the ipython shell, then the next time I  call a layout viewer, the one that was hanging finally closes and a new layout viewer window opens.

I have noticed that if I open ipython, specifying tk backend i.e. "ipython --pylab=tk" then the viewer window will eventually close after first python>"quit python", followed by two (2) clicks of thew red button. This works both in the 3.6.3  main env and the 2.7 env.  I say eventually because it usually fails (i.e. beachball) on the first invocation of "ipython --pylab=tk", but then starts to the work on the second or third.

egholami, 2018-02-19:
Use eclipse with pydev it would work fine!

heitzmann, 2018-07-20:
Unfortunately I can't work on this problem as I don't have access to MacOS. What I know is that Tk support on the Mac is not very good, so it could be a Tk/ipython issue.

An option is to avoid ``LayoutViewer`` is use an external viewer (I recommend [KLayout](https://www.klayout.de/)).

closed 2020-08-02 by heitzmann