heitzmann/gdspy #34: tkinter dependency

trezitorul, 2017-12-21:
If tkinter is not present on the system then LayoutViewer fails to import at all:

>>> import gdspy
>>> gdspy.LayoutViewer()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'LayoutViewer'

Just thought I would let you know that it fails without indicating that a dependency is missing.

heitzmann, 2017-12-21:
@trezitorul When Tkinter is not found, importing gdspy should issue a warning ("[GDSPY] LayoutViewer not available"). I don't see that in your code... have you disabled warnings?

Anyway, Tkinter is part of the python standard library, so I'm not sure it should be listed as a dependency.

mpinihin, 2018-02-26:
Hello, 
I have this same problem, I've been using gdspy on my windows computer with python 3 with no problems, but I just installed the gdspy module on my laptop running ubuntu using pip and I get the same error message as OP.

heitzmann, 2018-02-27:
It seems i'm wrong about tkinter being parto of the standard library: https://docs.python.org/3/library/tkinter.html

Still there should be a warning message if tkinter is not foud: "[GDSPY] LayoutViewer not available"
Do you see this warning @mpinihin?

mpinihin, 2018-02-27:
I was not getting that warning message. On my windows computer I installed python though the Anaconda distribution and it seems like tkinter was included there, but on my laptop python 3.6.3 came standard with ubuntu 17.10 and tkinter was not included. I fixed the problem by using  ``sudo apt-get install python3-tk``

heitzmann, 2018-02-27:
I've added Tkinter as an optional dependency. I believe this fixes the issue, but feel free to reopen it if you think there's anything else.

closed 2018-02-27 by heitzmann