heitzmann/gdspy #36: Output Error while generating large number of fine slices

srikanth1986, 2017-12-28:
I need to make fine slices of the polygons for my application and I wrote a script for the same. While the script itself doesn't throw any error, I see that polygons generated after slice operation are shifted from the original in a specific pattern. Here is original polygons superimposed.

[image]

The slices are aligned with the original in the left and right edges, but shifted down in the middle of the polygon.
Here is the same output zoomed in with original polygon highlighted in white.

[image]

And here is one of the slices highlighted in white.

[image]

This is the code that generates this result.
`#!/usr/bin/python
import numpy 
import gdspy

def frange(start, stop, step):
	i = start
	while i < stop:
		yield i
		i += step

poly_cell = gdspy.Cell('POLYGONS')
points = [(10.43631016, -35.44995495), (20.43478711, -35.27543088), (20.41733471, -34.27558319), (10.41885775, -34.45010725)]

polylist = []

poly2 = gdspy.Polygon(points, 0)
poly_cell.add(poly2)
polylist.append(poly2)

bbox = poly_cell.get_bounding_box()
print (bbox)
sl = []

for i in frange(bbox[0][0]+0.005, bbox[1][0]+0.005, .01):
	sl.append(i)

result = gdspy.slice(polylist, (sl), 0)
poly_cell.add(result)
gdspy.LayoutViewer()
`

heitzmann, 2017-12-28:
Which version of gdspy are you using? I'm guessing 1.2 or latter, because this looks like a precision error.
`slice` has a keyword argument `precision` that is set to 0.001 by default, which means all polygon vertices are aligned to a grid with 1e-3 spacing.

Try to run with:
```python
result = gdspy.slice(polylist, (sl), 0, precision=1e-6)
```
to see if that solves the problem.

srikanth1986, 2017-12-28:
Yes, That fixes it. I wasn't sure what the precision parameter was for.
Maybe you can add that to the description here
http://gdspy.readthedocs.io/en/latest/geometry.html#slice
And thanks for the super quick reply!

heitzmann, 2017-12-28:
Thanks! I missed that in the docs! Fixed in 5ca48f0.

srikanth1986, 2017-12-28:
Although this has fixed the problem for me, I do have some concern about
this solution. One, I was slicing at 0.01 intervals while the default
precision was 0.001. The second being that, what you say should have
resulted in an abrupt shift like a step function, but if you look at the
right half of the images, the slices catch up with original in a rather
smooth curve. Also I noticed the same thing happening in rectangles half as
long as this where the slices at the left and right are aligned while those
in the middle are not. Grid snapping doesn't explain this in my opinion.

heitzmann, 2017-12-28:
Indeed... that is an artifact of the combined effects of the grid snapping and the way slicing is performed: from left to right, one slice at a time.
To solve the issue I believe the `_chop` function in the C-extension should perform all chops (from the list of `positions` in a single run).
I'll reopen the issue until this can be implemented.

heitzmann, 2017-12-29:
@srikanth1986 I believe changes in https://github.com/heitzmann/gdspy/commit/9a951ae8939bfb8acb7d20efa6e9e5c6437aa5b1 solve this issue. I've tested your example and it works as expected for any precision setting.
Please let me know if you find any other issue.

srikanth1986, 2018-01-02:
> 9a951ae

Hi Lucas, How do I get this update for python on windows? I tried both
pip install --upgrade gdspy
and then downloaded the files and ran
python setup.py install

I still see the old output.

heitzmann, 2018-01-02:
@srikanth1986 You can get the latest Windows builds from AppVeyor: https://ci.appveyor.com/project/heitzmann/gdspy
Just select the environment corresponding to your python version and platform, then go to the Artifacts tab and download the whl file.

srikanth1986, 2018-01-02:
> t/heitzmann/gdspy

That works. Thank you very much!

closed 2018-01-02 by heitzmann