heitzmann/gdspy #39: Copy cells layers

acbg, 2018-02-07:
Having a cell with different layers and references, is it possible to copy/translate one o multiple layers?

I have a "main cell" which is a set of multiple cells and cell references.  These contain different elements in seven different layers.  I wonder if it's possible to make a copy/translation of the whole "main cell", with the same set of cells and references, but just containing one of the layers.

acbg, 2018-02-07:
A simpler way to ask the question: Is it possible to extract only one of the layers from a cell (composed by multiple sub-cells), and operate with the whole layer?

neloyo, 2018-02-07:
Is it possible to do Mirror copy?? cell, Polygon

acbg, 2018-02-07:
I guess it should. My question is if it's possible to mirror/copy just one of the layers present in a cell.

heitzmann, 2018-02-07:
For now this can only be accomplished more or less manually. Here's an example:
```python
import gdspy

main = gdspy.Cell('main')

for i in range(3):
    ref_cell = gdspy.Cell('ref' + str(i))
    ref_cell.add(gdspy.Round((i, i), 0.5, number_of_points=4 + i, layer=i))
    main.add(gdspy.CellReference(ref_cell, (0, 0)))
    main.add(gdspy.Round((0, -1), 0.5, number_of_points=3 + i, layer=i))

# Create a new cell and manually copy contents
copy = gdspy.Cell('main-copy')

# Get all polygons and keep only those in layer 2, dataset 0
# Any transformation can be applied to these polygons
polys = main.get_polygons(True, 0)
if (2, 0) in polys:
    for poly in polys[(2, 0)]:
        poly[:, 0] += 2  # translate in x
        poly[:, 1] *= -1  # mirror in y
    copy.add(gdspy.PolygonSet(polys[(2, 0)], layer=2))

# Do the same for all references
# Unfortunately this has to be done through `elements`
for el in main.elements:
    if isinstance(el, gdspy.CellReference):
        ref_copy = gdspy.Cell(el.ref_cell.name + '-copy')
        polys = el.ref_cell.get_polygons(True, 0)
        if (2, 0) in polys:
            ref_copy.add(gdspy.PolygonSet(polys[(2, 0)], layer=2))
            # Now add the reference (note that we apply the same
            # transformations to the reference as to the polygons in the
            # main cell)
            origin = list(el.origin)
            origin[0] += 2
            copy.add(
                gdspy.CellReference(ref_copy, origin, el.rotation,
                                    el.magnification, not el.x_reflection))
    elif isinstance(el, gdspy.CellArray):
        # Analogous to CellReference
        pass

gdspy.LayoutViewer()
```

acbg, 2018-02-12:
Thanks for the quick answer Lucas, it's exactly what I was looking for!

closed 2018-02-12 by acbg