heitzmann/gdspy #40: Layer Isolation and cells Boolean operation

egholami, 2018-02-19:
Hi
 I wanted to export a single layer from the gds file but I couldn't find the way for it so I have added a method using the flatten method to do it for me, but here is the problem when the file is slightly big(10MB) and I want to use my hand made method It becomes very time consuming so I wanted to ask how could I do Boolean operation with two cells from each other?
I am sure that what I have done could become more efficient using the appropriate library function but I don't know how.

heitzmann, 2018-02-19:
@egholami I'm not sure I understand what you're trying to do, but probably `cell.get_polygons(by_spec=True)` should help.

If you can post a minimal example, it might be easier to help you.

egholami, 2018-02-19:
**This is the Trim method that I add to Cell class:****
    def Trim   (self,
                mash):
        poly_dic = self.get_polygons(True)
        self.elements = []
       for ld in poly_dic.keys():
            if ld[0]==mash:
                self.add(PolygonSet( poly_dic[ld],ld[0],   verbose=False))
        return self

**It is working for Cell bolean Operation I am using Get_polygons but it is very slow:**

layli=['EBL','Trenches','Mesa']
for i in range(1,4):
    globals()['sepLayer%s'%i]=Ref_cell.copy('{}'.format(layli[i-1]),  deep_copy=True)
    globals()['sepLayer%s'%i].Trim(mash=i)
    globals()['sepLayer%s'%i]=gdspy.fast_boolean(globals()['sepLayer%s'%i].get_polygons(False, None),Iqub0m,'not',layer=i)
    re_cellmodified.add(globals()['sepLayer%s'%i])   
**for one of my layers that have alot of points and elements it is very slow but it is working**
_**Now I have tried what you suggested **_

for i in range(1,4):
    globals()['sepLayer%s'%i]=flatten.copy('{}'.format(layli[i-1]),  deep_copy=True)
    globals()['sepLayer%s'%i].Trim(mash=i)
    globals()['sepLayer%s'%i]=gdspy.fast_boolean(globals()['sepLayer%s'%i].get_polygons(by_spec=True),Iqub0m,'not',layer=i)
    re_cellmodified.add(globals()['sepLayer%s'%i])   
    print(i)

It is giving me this error! 

Traceback (most recent call last):

  File "C:\Users\ehgh\AppData\Local\Continuum\anaconda3\lib\site-packages\gdspy\__init__.py", line 4274, in fast_boolean
    result = clipper.clip(polyA, polyB, operation, 1 / precision)
TypeError: 'numpy.int32' object does not support indexing

Is there any way to make Boolean Cell operation fast?

heitzmann, 2018-02-19:
@egholami `get_polygons(by_spec=True)` returns a dictionary, so you're a missing the key (similar to what you do with `poly_dic`. ALso, you probably don't need to use `flatten()` or `copy()`.

In any case, boolean operations are slow. The only way around it is to use fewer points... I find people often overshoot the precision they need in curved geometries. Another option would be to try to avoid using boolean operations at all by drawing your geometry some other way.

_Unrelated to the issue_: it's quite dangerous to be abusing `globals()` like that. Global variables are in general a bad design choice, let alone using `globals`. You'd be better of using your own dictionary.

egholami, 2018-02-20:
Thanks a lot! It was a pseudo code and you are right that using the globals() is a bad design choice I would keep it in mind. Using the way that you have explained there would be no need for hard copy.
Unfortunately we deal with some special kind of text and we could not use regular text  so I need my own designed text with opening. Because for some chemistry reasons some letters like "O" or "B" could not be used in closed form and need some opening.

closed 2018-02-28 by heitzmann