heitzmann/gdspy #42: how to load a gds file, flatten it, and extract all polygons

enuinc, 2018-03-16:
Hi

Great software and appear to have all the things I need. However, I can not figure out how to load a gds file (not very big, about 2MB), flatten its hierarchy, and extract all polygons in a given layer.

I am sorry for such an elementary question.

Thanks
enuinc

heitzmann, 2018-03-16:
Hello @enuinc,

Loading a GDSII file can be done with [`GdsLibrary`](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.GdsLibrary).
If your file has a single top cell, then something like this should work:

```python
import gdspy

lib = gdspy.GdsLibrary(infile='filename.gds')
main_cell = lib.top_cell()[0]
pol_dict = main_cell.get_polygons(by_spec=True)
polygons = pol_dict[(layer, datatype)]
```
Note that in the last line I assume you know the layer and datatype numbers you're after. If the polygons in the layer you want have different datatypes, you'll have to iterate over therm and append to the `polygons` list.

Also note there's no need to [`flatten`](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.Cell.flatten) the cell, because [`get_polygons`](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.Cell.get_polygons) already traverses the whole cell hierarchy.

egholami, 2018-03-22:
Hi
I guess it is
main_cell = lib.top_**level**()[0]

vandhanasiva, 2021-01-29:
By any chance can we plot these polygons?

heitzmann, 2021-01-29:
@vandhanasiva Each polygon in the list is simply an array of N vertices (a `numpy.ndarray` with shape `(N, 2)`), so any drawing library should be capable of drawing it. You can use, for example, [matplotlib](https://matplotlib.org/3.3.3/api/_as_gen/matplotlib.patches.Polygon.html), or look for something more efficient if you have too many of them.