heitzmann/gdspy #43: coordinate precision

enuinc, 2018-03-27:
Hi,

I have another question about the polygon coordinate precision extracted from GDS.

I am using the following code to get all polygons from a GDS file.

import gdspy
gdsii = gdspy.GdsLibrary()
gdsii.read_gds('out-file.gds')
main_cell = gdsii.top_level()[0]
pol_dict = main_cell.get_polygons(by_spec=True)
polygons = pol_dict[(41, 0)]

When I print out the values in polygons, I saw the following

[array([[2.16, 1.82],
       [2.16, 0.98],
       [1.82, 0.98],
       [1.82, 0.54],
       [2.26, 0.54],
       [2.26, 0.76],
       [2.34, 0.76],
       [2.34, 1.82]]), 
....

It seems I have to multiply 1000 to get the integer coordinates. Is this as expected?

I don't mind to multiply 1000 but I thought all numbers in GDS file is integer

Also, if I want to draw some polygons to output them to a GDS file, should I divide my coordinates by 1000?

Thanks
enuinc

enuinc, 2018-03-27:
I used another GDS reader and can confirm that the data is stored in the integer. It seems gdspy multiplies whatever values stored in GDS file with the so called DBUperUU, which is 0.001 in my case

I would like to confirm if my understand is correct or not.

Thanks
enuinc

heitzmann, 2018-03-27:
@enuinc That's correct, Gdspy presents data in *user units*, not the integers written in the GDSII file.

If you need to change the units when you import a GDSII file (because you are using different *user units* in different files, for example), it's easier to use the `unit` argument in [GdsLibrary.read_gds](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.GdsLibrary.read_gds).

You might also be interested in [this specification](http://boolean.klaasholwerda.nl/interface/bnf/gdsformat.html#rec_units).

enuinc, 2018-03-27:
@heitzmann Thanks for the reply. I have two additional questions
1. Is there a function in gdspy to give me the dbu and uu value in a gds file? 
2. In addition, can gdspy output the polygons in dbu instead uu?

Thanks
enuinc

heitzmann, 2018-03-27:
@enuinc 

1. Not at this moment, but it should be somewhat easy to implement...
2. If you know the size of *dbu* in meters, passing that value to `unit` will make all coordinates will be in *dbu* (nevertheless, `datatype` will still be `float`, not `int`)

Is there a specific use case you're thinking of?

enuinc, 2018-03-27:
@heitzmann 

I have a code that take the GDS coordinate and convert them to physical unit when needed. it will be quite a bit work to change various places in the code to take the physical unit directly. right now, it is fine as I can just assume I always work for nm unit and make a hard-coded conversion. 

Thanks
enuinc

heitzmann, 2018-04-14:
@enuinc In https://github.com/heitzmann/gdspy/commit/da2e56bc730dced87e0da867813a69e083b4dd47, [`GdsLibarry.read_gds`](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.GdsLibrary.read_gds) adds support to `'import'` the units from the GDSII file. `GdsLibrary.precision` is the file's *dbu* and `GdsLibrary.unit` is *dbu/uu*.

closed 2018-04-14 by heitzmann