heitzmann/gdspy #46: Nested Cells

smartalecH, 2018-04-06:
Is it possible to nest cells using gdspy? I've poured over the documentation/examples but haven't found any explicit reference to nesting or cell hierarchy.

I'm building a library of function components that return cells. I would like more complex cells to call simpler cell routines. The GDS format obviously works really well with this recursive framework.

heitzmann, 2018-04-06:
@smartalecH Indeed hierarchical structures is one of the strong points of GDSII. You can nest cells in Gdspy using [`CellReference`](http://gdspy.readthedocs.io/en/latest/library.html#cellreference) and [`CellArray`](http://gdspy.readthedocs.io/en/latest/library.html#cellarray).

The [tutorial](http://gdspy.readthedocs.io/en/latest/tutorial.html#tutorial) has examples of both.

closed 2018-04-06 by heitzmann

horror-vacui, 2021-06-08:
The link to the tutorial is broken now. I believe this has the [examples](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#references) now.