heitzmann/gdspy #52: GDS Timestamps

olegsyrel, 2018-06-03:
The present implementation seems to use current date and time for GDS cell and library timestamps. As a result, GDS files created from the same layout at different points in time will have different binary headers. While appropriate in most situations, this behaviour complicates some use cases that involve binary comparison of GDS files (e.g. regression testing).

I would like to propose to add an optional `timestamp=None` argument to `Cell.to_gds()`, `GdsLibrary.write_gds()`, and `GdsWriter()` constructor. If set to `None` (the default), the functions would use the current date and time, otherwise the user-supplied timestamp would be used.

heitzmann, 2018-06-04:
Sounds good. I'll get to it as soon as I have some time from work.

rosivda, 2018-06-07:
@olegsyrel  Please note that ```gdspy``` contains a facility called ```gdsii_hash(filename, engine=None)``` that helps compare GDSII binary files while ignoring any timestamp records. We use this function to store hashes of original binaries. Subsequently, we test for regression by regenerating the binaries and comparing their hashes to the original stored values.

olegsyrel, 2018-06-08:
@rosivda Thanks for the pointer! The `gdsii_hash` function will do the trick for me. I don't think it existed in v1.1, which I was using until recently; I guess it'd help my case to RTFM once in a while :)

heitzmann, 2018-06-15:
Fixed in https://github.com/heitzmann/gdspy/commit/34b06faf538271c55522edea76bbfe9f16097ed7

closed 2018-06-15 by heitzmann