heitzmann/gdspy #57: Adding a Cell directly within a Cell (without using a CellReference) causes "missing ENDSTR record" error when trying to open the GDS file that is written out

peterxsu, 2018-07-09:
Calling Cell.add(<Cell>) works fine with the LayoutViewer, but when the layout is then written to a GDS file, trying to open it up in Clewin 5 causes a "missing ENDSTR record" error. Trying to open it up in KLayout also causes an error.

Suggested fix is that trying to add a Cell directly to another Cell just inserts a CellReference at position (0,0), or something similar. Or maybe just throw an error? I'm sure trying to get the file to write correctly would take more time than either of these other two options.

heitzmann, 2018-07-16:
@peterxsu This is more an issue of the documentation than error-checking. Currently we don't do error-checking in the user input — the user is responsible for passing the correct arguments to the library.

I've fixed the documentation for ``Cell.add`` accordingly in 08f6e5a

harrywang2018, 2018-07-18:
suggest to update "def add()" in the "calss Cell" as below, and then cell be added directly in the cell

    def add(self, element,*args, **kwargs):
        if isinstance(element, list):
            for e in element:
                if isinstance(e, Cell):
                    self.elements.append(CellReference(e, *args, **kwargs))
                elif isinstance(e, Label):
                    self.labels.append(e)
                else:
                    self.elements.append(e)
        else:
            if isinstance(element, Cell):
                self.elements.append(CellReference(element, *args, **kwargs))
            elif isinstance(element, Label):
                self.labels.append(element)
            else:
                self.elements.append(element)
        self._bb_valid = False
        return self

heitzmann, 2018-07-18:
@harrywang2018 That would implicitly create a ``CellReference`` which can be confusing... "Explicit is better than implicit."