heitzmann/gdspy #58: write_gds(outfile,cells) can't export the CellReference in the cells

harrywang2018, 2018-07-18:
I use conder python 3.6 and gdspy 1.3.1, I found if I set cells=[topcell], write_gds can't export the CellReference in the topcell, I must set cells=None, but the other cell will be exported too.

so I update the gdspy __init__ code as below, for your reference.
I add one more function.
def CellList(cell,celllist=[],First=True):
    if isinstance(cell,Cell):
        if First==True: 
            celllist.append(cell)

        for r in cell.elements:
            if isinstance(r,CellReference):
                celllist.append(r.ref_cell)
                CellList(r.ref_cell,celllist=celllist,First=False)
        return celllist
    else:
        return None

and also, I update def wirte_gds() as below:
def write_gds(outfile,
              cells=None,
              name='library',
              unit=1.0e-6,
              precision=1.0e-9):
    current_library.name = name
    current_library.unit = unit
    current_library.precision = precision
#    current_library.write_gds(outfile, cells)
    
    totalcell=[]
    if isinstance(cells,Cell):
        totalcell=cellList(cells,celllist=[])
        current_library.write_gds(outfile, totalcell)
    elif isinstance(cells,list):
        for cell in cells:
            totalcell=totalcell+cellList(cell,celllist=[])
        totalcell=set(totalcell)
        current_library.write_gds(outfile, totalcell)

and then, if my code is gdspy.write_gds('a.gds',topcell), all the cellreference in the topcell can be exported  successfully.

harrywang2018, 2018-07-18:
still many bugs......only can set cells=None.....

heitzmann, 2018-07-18:
@harrywang2018 That is the **expected** behaviour. Please read the note in the documentation for [``write_gds``](http://gdspy.readthedocs.io/en/latest/library.html#gdspy.GdsLibrary.write_gds):

> Only the specified cells are written. The user is responsible for ensuring all cell dependencies are satisfied.

closed 2018-07-18 by heitzmann