heitzmann/gdspy #61: Path widths not conserved for number_of_paths higher than 1

dwaardenburg, 2018-07-31:
I noticed that when I use a number_of_paths higher than 1 when I change the path distance between the paths this does not conserve the widths of the paths it draws. Only the widths of the paths at the beginning and the end of the tapered section. I would say that these two parameters could be simultaneously the width of the line and would strongly suggest this should be fixed, as for the effect can be difficult to observe but lead to errors none the less.

heitzmann, 2018-07-31:
In the case of a change in distance between paths, the the width, as measured by gdspy, remains constant in the direction perpendicular to the segment. The problem is that the direction of the segment and the direction of each path are not parallel in this case, so the width of each path (measured orthogonally to it) will be different.

I can't see a way to solve this divergence in the current implementation, because it would require modifying the previous path section, which could be anything (including non existent).

Any suggestions?

dwaardenburg, 2018-07-31:
After having thought about it for a little longer it would indeed be a very difficult thing to solve. However, I still think there is an definite need for solving this as the code suggest that the correct width is obeyed. That is not the case. 

So far I have a couple of suggestions:
 - I would think we only need to know the direction and the width of the old path and the new path to solve the problem with a boolean operation.  This would then allow some sort of fix depending on what dimensions are most critical (I would say that would be width as length of paths can be fixed in other locations). The turning of multiple paths does not have this problem 
[image] This however makes conserving math lengths very difficult, but conserves path widths if I see it correctly. None the less, I would suggest that we look for the solution in that code.
- The more simple thing is to just warn the user that the widths/lengths are not conserved.

I am currently working on a solution for my device by just drawing each path separately and fixing the gaps in the corners by just making both polygons longer on the side that is coming short.

If I have more information I will come back to you...

heitzmann, 2019-02-10:
@dwaardenburg The new classes [SimplePath](https://gdspy.readthedocs.io/en/next/gettingstarted.html#simple-paths) and [LazyPath](https://gdspy.readthedocs.io/en/next/gettingstarted.html#lazy-paths) correctly support the behavior you described. If you're still interested, you can take a look into the next branch. It shouldn't be long before a proper release, though.

dwaardenburg, 2019-02-10:
Wow, that looks a lot more spectacular than the name suggests. I must say I will need to get a bit back in the game though to test them out a little, currently I am just using custom functions where possible. Just today I would have loved to use the SimplePath for a structure that I was drawing. I thus far have only one comment after a short look. The names seem somewhat arbitrary, maybe call SimplePath -> FlexPath and LazyPath -> RobustPath. Those would quite better describe what the classes are designed for.

heitzmann, 2019-02-10:
Makes sense... The names I chose are based on an implementation point of view, not the user's. Thanks!

closed 2019-04-26 by heitzmann