heitzmann/gdspy #62: Is there a way to overwrite duplicate cells when using extract?

peterxsu, 2018-08-13:
Trying to import a bunch of similar designs from externally made GDS files that have identical subcells.

I imagine the easy way to do this would be to add an overwrite_duplicate=False variable to extract and pass it directly to add.

heitzmann, 2018-08-14:
@peterxsu
It's not a simple matter of duplicate names that define identical subcells. Cells could have the same name and different contents. It could be dangerous for users to fall into this trap without knowing it.

In your case, if you *know* that cells with the same name are identical, the following code should give you an idea of what you have to do to merge the files:

```python
import gdspy

def create_gdsii(n):
    fname = 'file{}.gds'.format(n)

    cell1 = gdspy.Cell('REFERENCED', True)
    cell1.add(gdspy.Rectangle((0, -0.5), (1, 0.5)))
    cell1.add(gdspy.Label('Cell from {} '.format(fname), (0, 0), 'e'))

    cell2 = gdspy.Cell('COPY{}'.format(n), True)
    cell2.add(gdspy.CellArray(cell1, 1, 1 + n, (0, 2)))

    lib = gdspy.GdsLibrary()
    lib.add([cell1, cell2])
    lib.write_gds(fname)
    return fname

# Create 2 example files
file1 = create_gdsii(1)
file2 = create_gdsii(2)

# Load first file
lib1 = gdspy.GdsLibrary(infile=file1)
# Extract all cells:
for cell in lib1.top_level():
    lib1.extract(cell)

# Load second file
lib2 = gdspy.GdsLibrary(infile=file2)

# Change references based on cell names
# WARNING: this is no guarantee that the cells are identical!
for cell in lib2.cell_dict.values():
    for element in cell.elements:
        if (isinstance(element, gdspy.CellArray) or
                isinstance(element, gdspy.CellReference)):
            name = element.ref_cell.name
            if name in gdspy.current_library.cell_dict:
                element.ref_cell = gdspy.current_library.cell_dict[name]

# Manually extract all cells skipping duplicate names
for name, cell in lib2.cell_dict.items():
    if name not in gdspy.current_library.cell_dict:
        gdspy.current_library.add(cell)

gdspy.LayoutViewer()
```
Note that, if you are using several files, I'd advise against using the global `gdspy.current_library`. Instead, you can work with a `merged_lib` as follows:

```python
import gdspy

def create_gdsii(n):
    fname = 'file{}.gds'.format(n)

    cell1 = gdspy.Cell('REFERENCED', True)
    cell1.add(gdspy.Rectangle((0, -0.5), (1, 0.5)))
    cell1.add(gdspy.Label('Cell from {} '.format(fname), (0, 0), 'e'))

    cell2 = gdspy.Cell('COPY{}'.format(n), True)
    cell2.add(gdspy.CellArray(cell1, 1, 1 + n, (0, 2)))

    lib = gdspy.GdsLibrary()
    lib.add([cell1, cell2])
    lib.write_gds(fname)
    return fname

# Create 2 example files
file1 = create_gdsii(1)
file2 = create_gdsii(2)

# Load first file into merged library
merged_lib = gdspy.GdsLibrary(infile=file1)

# Load second file
lib2 = gdspy.GdsLibrary(infile=file2)

# Change references based on cell names
# WARNING: this is no guarantee that the cells are identical!
for cell in lib2.cell_dict.values():
    for element in cell.elements:
        if (isinstance(element, gdspy.CellArray) or
                isinstance(element, gdspy.CellReference)):
            name = element.ref_cell.name
            if name in merged_lib.cell_dict:
                element.ref_cell = merged_lib.cell_dict[name]

# Manually extract all cells skipping duplicate names
for name, cell in lib2.cell_dict.items():
    if name not in merged_lib.cell_dict:
        merged_lib.add(cell)

gdspy.LayoutViewer(merged_lib)
```

peterxsu, 2018-09-01:
Thanks @heitzmann . Sorry for not seeing this earlier.

Is there a reason that add has the overwrite_duplicate functionality then, and that's not considered too dangerous to give as an option?

closed 2018-11-22 by heitzmann