heitzmann/gdspy #63: Adding metadata to gds or cells / Saving cell-libraries in hard-drive for future use

stringflamer, 2018-08-15:
Hi Lucas,

Two questions:
1) Is there a way to add metadata to cells or gds? Metadata such as 'name', 'creator', 'version' ...
2) I have a bunch of common external .gds from manufacturers that i need to import for every .gds i draw. Is there a way to import just once, then save them in a file in hard-drive, and simply just import that saved file in every one of my gdspy scripts (avoid importing the external gds one-by-one everytime I write a new gdspy script)?

Thanks,
Lawrence

heitzmann, 2018-08-16:
Hi Lawrence!

1. Not in the standard GDSII format. You could add this information as the library name (I forget now what the maximum size of the name can be, but it should be more than enough).

2. We would have to come up with a "gdspy" file format for more efficient saving and loading data. There's nothing like that in the current versions, but it could be done. It's no easy task, though...

stringflamer, 2018-08-16:
Now i'm thinking about this, maybe the metadata can be added into the "gdspy" file, meaning combining 1. & 2.. This should be a good direction for gdspy to explore.

closed 2018-11-22 by heitzmann