heitzmann/gdspy #64: Behaviour of PolygonSet.fillet on non-trivial polygon

JBraumueller, 2018-08-20:
I created a new PolygonSet by applying
`shape=gdspy.fast_boolean(Rectangle, Circle, 'not')`
where Rectangle and Circle are partially overlapping. After that I merged all resulting polygons via `merged=gdspy.fast_boolean(shape, None, 'or')`
and applied the fillet method
`merged.fillet(20, 100)`.
In the resulting shape, corners along the overlap path of the original shapes are not rounded accordingly or an erroneous shape is being generated.

heitzmann, 2018-08-21:
@JBraumueller You probably want to avoid fracturing before the fillet operation. Both `shape` and `merged` will be fractured if you don't specify otherwise. To do so you have to set the `max_points` argument in [`fast_boolean`](https://gdspy.readthedocs.io/en/stable/geometry.html#gdspy.fast_boolean) to something sufficiently large (or zero to disable checking completely).

In that case you probably don't need to manually merge the result from the first operation:

```python
shape = gdspy.fast_boolean(Rectangle, Circle, 'not', max_points=0)
shape.fillet(20, 100)
```

Note that if the circle is inside the rectangle (`shape` has a hole) `fillet` is probably give the wrong result because of the connection between the hole and the outside boundary. In that case, you'll probably want to run `fillet` before the boolean operation.

JBraumueller, 2018-08-21:
Thank you very much for your answer.
I was using max_points=0 in the merge and I just tested using max_point=0 as you suggested without the additional merge. In both cases, the fillet works at the corners of the rectangle, but is not applied to the cutting-edge between rectangle and circle. The circle was not within the rectangle but only cuts out some section.
```
rectangle = gdspy.Rectangle([0,0], [50, 50])
circle = gdspy.Round([50+25], 25], 30)
shape = gdspy.fast_boolean(rectangle, circle, 'not', precision=0.01, max_points=0)
shape.fillet(20, 100)
```
gives me a non-rounded corner at the region where the circle was cut out from the rectangle.
Thank you again!

heitzmann, 2018-08-22:
That's expected. Unfortunately the `fillet` function only works at vertices whose edges are long enough to touch a circle with the radius specified at both tangent points. That's probably not the case at the intersection of the rectangle and the circle, since the latter is composed of very small edges.

JBraumueller, 2018-08-22:
Ok, I see. Thank you very much.

closed 2018-08-22 by JBraumueller