heitzmann/gdspy #69: [GDSPY] cell named "name of cell" already present in library

totyped, 2018-10-18:
Hi ! 

I am using Jupyter-Notebook on Ubuntu to work with gdspy. When I run it for a second time, I always get: 

> [GDSPY] cell named "name of cell" already present in library.

So, my work-around is either to restart the jupyter kernel or rename the cells, but both methods are quite annoying if you want to quickly test things. 

How can I reinitialize the library every time I run the code ?

heitzmann, 2018-10-18:
You can either use the argument `exclude_from_current=True` when creating a [cell](https://gdspy.readthedocs.io/en/latest/library.html#gdspy.Cell) or reset the [global library](https://gdspy.readthedocs.io/en/latest/library.html#gdspy.current_library) with `gdspy.current_library = gdspy.GdsLibrary()`

totyped, 2018-10-18:
Thanks a lot ! This worked ! :))