heitzmann/gdspy #70: Individual filleting of each corner in fillet function broken

PSpring, 2018-10-30:
Trying to fillet each corner of a shape individually by giving the fillet function a list of fillet values matching the number of corners of the polygon has been broken since update .

Doing so now will result in errors of the kind: 

`
Traceback (most recent call last):
  File "...\gdspy\__init__.py", line 448, in fillet
    l = radius * tt[ii]
TypeError: can't multiply sequence by non-int of type 'numpy.float64'
`

This functionality seems to have been lost when the polygon class was cleaned up and made to inherit from the polygotset class in update: Beforehand, the fillet function in the polygon class had a section like: , whereas fillet in the polygonset class went like:

So when Polygon inherited fillet from Polygonset, the individual corner filleting functionality to was lost.

Hopefully, simply reinserting these lines into the Polygonset class should be enough to restore this really useful functionality.

heitzmann, 2018-10-30:
Thanks @PSpring! I believe 958e31e35214cc456e8996284e94c77fef14ca01 fixes this bug.

PSpring, 2018-10-30:
No problem! Thanks for the super quick fix!

closed 2018-11-22 by heitzmann