heitzmann/gdspy #72: Wafer Background

totyped, 2018-11-05:
Is it possible to add a layer where you just have the border of a 4inch (100mm) wafer, just that one can better estimate the size of the design ? 

Thanks!

heitzmann, 2018-11-06:
Sure, you can simply draw a `Round` with the proper radius. Just be aware that the default number of points will be too large for a large radius, sou you'll have to decrease it to something more reasonable.

For example, using a default unit of 1 μm:

```python
import gdspy

chip = gdspy.Cell('CHIP')
chip.add(gdspy.Rectangle((-4_000, -4_000), (4_000, 4_000), layer=1))

main = gdspy.Cell('MAIN')
main.add(gdspy.Round((0, 0), 50_000, number_of_points=128))
main.add(gdspy.CellArray(chip, 5, 5, (10_000, 10_000), (-20_000, -20_000)))

gdspy.LayoutViewer()
```

heitzmann, 2018-11-09:
Sure, you can use [`Path.arc`](https://gdspy.readthedocs.io/en/latest/geometry.html#gdspy.Path.arc) with an appropriate radius and width, or a `Round` with an appropriate `ineer_radius`. Both will give oyu the same result, but I find `Round` easier to use.

closed 2018-11-22 by heitzmann