heitzmann/gdspy #73: Function To Center Design At Origin

totyped, 2018-11-05:
Hi ! Is it possible to create a function, which centers the current design at (0,0) ?

I tried this:
```
def center_object(cell):
    centered_cell = gdspy.Cell('centered_cell')
    p0, p1 = cell.get_bounding_box()
    centered_cell.add(gdspy.Polygon(cell.get_polygons()).translate(-p0[0]/2, -p1[0]/2))
    gdspy.LayoutViewer()
    return centered_cell
```

heitzmann, 2018-11-06:
Hello @totyped!

Instead of translating the polygons, it is easier (and safer) to move the reference origin:

```python
import gdspy

def center_reference(reference):
    bbox = reference.get_bounding_box()
    # Move the reference origin to put the center of its bounding box at (0, 0).
    reference.origin -= 0.5 * bbox.sum(axis=0)
    # The reference is modified in place.
    # The same object is returned just for convenience.
    return reference

top_cell = gdspy.Cell('top_cell')

#Sample GDS
unit_cell = gdspy.Cell('unit_cell')
w = 20
N = 2
size = 2 * w * N
inner_rec = gdspy.Rectangle((-w / 2 - size / 2, -w / 2 - size / 2),
                            (w / 2 - size / 2, w / 2 - size / 2))
outer_rec = gdspy.Rectangle((-w - size / 2, -w - size / 2),
                            (w - size / 2, w - size / 2))
unit_cell.add(gdspy.fast_boolean(outer_rec, inner_rec, 'not', layer=0))
rec_array = gdspy.CellArray(unit_cell, N, N, (2 * w, 2 * w))

# Center CellReference or CellArray
center_reference(rec_array)

# Add to Top Cell
top_cell.add(rec_array)

# Visualize
gdspy.LayoutViewer()
```

totyped, 2018-11-06:
Wow. This works really nicely !
I didn't know that you could modify the reference in place. Very cool.

closed 2018-11-22 by heitzmann