heitzmann/gdspy #74: When dealing with circles the output points are reduced by library

aayushraina, 2018-11-09:
When dealing with circles, I am converting it to set of 32 points. However gdspy is reducing the number of points in the circle. Is it #clip function which is doing it or #fracture

        poly3 = gdspy.fast_boolean(my_list1, my_list2, operation, max_points=5000, layer=776)

Input

[image]

Output

[image]

[InputandOutput.zip](https://github.com/heitzmann/gdspy/files/2564760/InputandOutput.zip)

heitzmann, 2018-11-09:
@aayushraina Your geometry doesn't look fractured, so I believe it's the `clip` (inside [`fast_boolean`](https://gdspy.readthedocs.io/en/latest/geometry.html#gdspy.fast_boolean)) doing the simplification. It is probably related to the `precision` parameter. Have you tried changing it from the default 1e-3 to something like 1e-6?

The library that performs the clipping operation works only with integer coordinates, so the geometry is scaled by dividing by this factor, then all coordinates are converted to integer, the operation is performed, and finally the resulting coordinates are multiplied by the same factor. Depending on your drawing scale (default: *s = 1 µm*) and the precision you set (default: *1e-3 * s = 1 nm*), some of the coordinates might lie on top of each other and end up merged.

aayushraina, 2018-11-12:
Thanks . I will check and update. For now I am closing the issue.

closed 2018-11-12 by aayushraina