heitzmann/gdspy #75: Is there a way to offset all polygons by a certain number ?

totyped, 2018-11-09:
I would like to offset all polygons in my design by a certain number, so what I tried was the following:

```#To reset the libary every time you run the code
gdspy.current_library = gdspy.GdsLibrary()

top_cell = gdspy.Cell('top_cell')
top_cell.add(gdspy.Rectangle((18, 1), (22, 2), 2))
top_cell.add(gdspy.Rectangle((19, 0), (20, 2), 2))
top_cell.add(gdspy.Round((27, 2), 2, layer=2))

#Here I try to do the offset:
envelope_cell = gdspy.Cell('envelope_cell')
envelope_cell.add(gdspy.offset(top_cell.get_polygons(), 2, join_first=True, layer=3))

gdspy.LayoutViewer()
```

This is how it looks like without offset:
[image]

And with offset:
[image]

As you can see the offset is applied generally to the complete outline, but I wonder if it is possible to apply it to the individual polygons? 

Thanks a lot !

heitzmann, 2018-11-09:
Yes, you have to use `offset` in each element separately:

```python
import gdspy

top_cell = gdspy.Cell('top_cell')
top_cell.add(gdspy.Rectangle((18, 1), (22, 2), 2))
top_cell.add(gdspy.Rectangle((19, 0), (20, 2), 2))
top_cell.add(gdspy.Round((27, 2), 2, layer=2))

envelope_cell = gdspy.Cell('envelope_cell')
for element in top_cell.elements:
    envelope_cell.add(gdspy.offset(element, 2, join_first=True, layer=3))
```

heitzmann, 2018-11-12:
I believe it is possible, but not easy to implement. You'd have to test each envelope against all other with a Boolean function to check if either one is inside the other.

To compare a pair of polygons to see if one is inside the other you can join them with `or` and then `xor` the result with the largest (in area). If the result from the `xor` is empty, the smallest is completely inside the largest.

But beware that Boolean calls are expensive and your script will slow down for large numbers of envelopes.

closed 2018-11-22 by heitzmann