heitzmann/gdspy #79: Support GDS paths

gyger, 2018-12-05:
It would be nice if the Path datatype of GDS would be supported. [1] I am not sure if I just miss something, but if I create a path with gdspy, it is a Box (Boundary [2] ) in KLayout, while if I create a Path in Klayout or other tools, it seems to be a different structure type.

[1] http://www.artwork.com/gdsii/gdsii/page4.htm
[2] http://www.artwork.com/gdsii/gdsii/page3.htm

shermelin, 2018-12-08:
Thanks for the nice job on gdspy!

I would like to second that it would be nice to have the ability to write gdsii path, since it can turn out useful for some fabrication. I guess it could be an independent class, say gdspy.BasicPath or gdspy.GdsiiPath.

gyger, 2018-12-08:
Sorry for the late reply. It is used in some control elements in the GDS
for the Raith Ebeam software.

heitzmann, 2018-12-10:
I see. I'll try to have a first implementation by the next release, which should come early next year.

heitzmann, 2019-02-07:
I'm glad to say that Paths are fully supported in the next branch of gdspy if anyone would like to try it out. The documentation is already updated as well: https://gdspy.readthedocs.io/en/next/

The next release will definitely include these changes and should be ready before the next month, I hope.

closed 2019-02-07 by heitzmann