heitzmann/gdspy #80: Why this library over other gds libraries in python

gyger, 2018-12-13:
I think it would be useful to have a short comment in the README what are the advantage of this library over the other GDS libraries available in python, to help people to choose the right library for their project.

heitzmann, 2019-02-10:
Hi @gyger, thanks for the suggestion. To be honest, when I first started working on gdspy back in 2009 I did this kind of research. In fact, I created gdspy because I couldn't find any library that would work for my specific use case (creating huge amounts of simple polygons without running out of RAM).

However, I didn't keep up with what other libraries have to offer nowadays, so I'm not sure I can compare them to gdspy. I'd have to do some research and then continue doing so periodically to keep the comparison updated, but my time is quite short as it is... I barely have time to implement all the new features I would like to.

Maybe at some point I try doing something like your suggestion, but I can't promise anything in the near future. Of course I always welcome contributions in case you've started this comparison.

gyger, 2019-02-24:
I did not, I just chose yours because it was suggested/base of another
library Picwriter. I did not regret it. Thanks.

2019, 13:29:

closed 2020-08-02 by heitzmann