heitzmann/gdspy #82: gds_image replacement/workaround?

McCrearyD, 2019-03-08:
I noticed that there was a function "gds_image" where you could take the images created from the cells in the "LayoutViewer" and export them as images. Is this still possible through gdspy? Or is there a solution better than what there was before?

heitzmann, 2019-03-11:
@McCrearyD This function was removed because it was not the main purpose of gdspy to create images.
There's currently no replacement for it, but you can either take a screenshot from `LayoutViewer` or take a look into [the `draw` function](https://github.com/heitzmann/gdspy/blob/next/docs/makeimages.py#L16) in makeimages.py (this is a helper for generating the docs that creates images using PIL). It probably contains all the code you need to write your own version of `gds_image`.

McCrearyD, 2019-03-13:
I actually ended up making my own script to do so. It just generates a numpy array of a specific width, height, & depth and fills the slots in with their data types etc.

McCrearyD, 2019-03-13:
If anyone is interested, here is that script: https://gist.github.com/McCrearyD/f0d156ae7b14ee2bb8c51587e2c7adad

closed 2019-03-23 by McCrearyD