heitzmann/gdspy #85: What the parameter "dimensions" in cell.to_gds(multiplier) doc mean?

operatorheaven, 2019-04-17:
hi~I'm novice for this library.
I don't understand the number named "multiplier" ref what.
`multiplier (number) – A number that multiplies all dimensions written in the GDSII element`

And i have another puzzle that is there a way to strmOut cell(obj: cell) to ***.gds gdsii-format file like as `GdsLibrary.write_gds()` ?
Only can binary-format strmOut or make one gdslibrary including this cell meet my request?
Thanks a lot!

heitzmann, 2019-04-17:
Hi @operatorheaven!

`multiplier` isa number that multiplies all vertices coordinates. It is used because gdsii files store integer coordinates + physical unit + precision. As a user, you shouldn't have to set multiplier yourself. Simply add the geometry you want to a cell, and save the (default global) library using [`gdspy.write_gds(...)`](https://gdspy.readthedocs.io/en/stable/library.html#gdspy.write_gds).

If you are a beginner, please take a look at the [tutorial](https://gdspy.readthedocs.io/en/stable/tutorial.html#tutorial). I believe it will clarify your second question.

There is also the new [Getting Started](https://gdspy.readthedocs.io/en/next/gettingstarted.html) page in the making, which might be helpful to you.

operatorheaven, 2019-04-18:
> Hi @operatorheaven!
>
> `multiplier` is a number that multiplies all vertices coordinates. It is used because gdsii files store integer coordinates + physical unit + precision. As a user, you shouldn't [...]

Got it! Thanks very much!
So the value of `multiplier `would be counted automatically when we use `GdsLibrary.write_gds("...")` to generate gdsii-format file?
Is that right please!

Now I follow this code to convert cell object to gds file(*.gds):
```
tmp_lib = gdspy.GdsLibrary()
tmp_lib.add(cell)
tmp_lib.write_gds("output.gds")
```
Is there any problem or risk?

heitzmann, 2019-04-18:
Yes, that's correct! Your code is OK as well.

operatorheaven, 2019-04-18:
Thanks sooooo much!

closed 2019-04-18 by operatorheaven