heitzmann/gdspy #86: Output problem

nquimaso, 2019-04-30:
Hello, I'm new in python  and also in gdspy and
Trying to run the tutorial until the output part I obtain this error:

`gdspy.write_gds('tutorial.gds', unit=1.0e-6, precision=1.0e-9)
AttributeError: module 'gdspy' has no attribute 'write_gds'`

why?
-----
On the other hand, If I run the code a second time after execute it says:

`ValueError: [GDSPY] A cell named POLYGONS has already been created.`

And I have to reset the console to run it again.  The point is I don't see the cells in the variable space

heitzmann, 2019-04-30:
Hello @nquimaso,
Which version of gdspy are you running and what is your OS? The error about `write_gds` not existing is quite strange.

Your second issue is that gdspy keeps a record of all created cells in [`gdspy.current_library`](https://gdspy.readthedocs.io/en/stable/library.html#gdspy.current_library).

If you're running in a console, you have to reset this library before running the tutorial a second time with: `gdspy.current_library = gdspy.GdsLibrary()`.

nquimaso, 2019-04-30:
I'm `Using gdspy module version 1.0` in my spyder IDE, not console sorry, in windows 10.

heitzmann, 2019-04-30:
This is a very old version. Please upgrade to a more [current version](https://github.com/heitzmann/gdspy/releases/). The function `gds_write` did not exist in v1.0, neither did `GdsLibrary()`.

nquimaso, 2019-04-30:
Ok, I remember that I installed this version because the unique way to install the library was explained here: https://github.com/heitzmann/gdspy/issues/13 

Now, I've downloaded the wheel file gdspy-1.3.2-cp27-cp27m-win_amd64 (1).whl
but installing with pip: 
`\Users\Nahuel\Desktop\WinPython-64bit-3.5.2.2>pip install "gdspy-1.3.2-cp27-cp27m-win_amd64 (1).whl"
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
gdspy-1.3.2-cp27-cp27m-win_amd64 (1).whl is not a supported wheel on this platform.
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
Could not fetch URL https://pypi.org/simple/pip/: There was a problem confirming the ssl certificate: HTTPSConnectionPool(host='pypi.org', port=443): Max retries exceeded with url: /simple/pip/ (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available.")) - skipping`

I've added the folder `C:\Python\Scripts` to the path previously but it doesn't work. 

Can someone explain me how can I install this library?

heitzmann, 2019-04-30:
I don't use Windows, but I believe your best bet for installing Python and all dependencies easily is to use [Anaconda](https://www.anaconda.com/distribution/).

closed 2019-05-02 by nquimaso