heitzmann/gdspy #89: get_bounding_box returning numbers magnitudes off

McCrearyD, 2019-06-09:
For a gds cell with a max element at ~ 2000, cell.get_bounding_box() returns: [[-0.285  0.465]
 [ 1.04   3.635]].

heitzmann, 2019-06-10:
@McCrearyD Could you provide a minimal example showing the problem? I cannot replicate what you describe...

McCrearyD, 2019-06-10:
Load [this gds file](https://drive.google.com/file/d/14dFqfcMNYX5naQ-OBPA5Ykd6P8lDU5RC/view?usp=sharing) into gdspy and on the cell, call "get_bounding_box()". This will return seemingly random floating point numbers.

heitzmann, 2019-06-10:
@McCrearyD The GDSII file is loaded correctly.
What you're seeing are coordinates expressed in micrometers (default for GDSII files), and you probably wanted nanometers.
Please take a look at [units in GDSII](https://gdspy.readthedocs.io/en/stable/gettingstarted.html#units-in-gdsii) and [`GdsLibrary`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary) for more information.

In short, you can get load this library in nanometers by using:
```python
import gdspy

gds = gdspy.GdsLibrary(infile="test.gds", unit=1e-9, precision=1e-12, units='convert')
c = gds.top_level()[0]
print(c.get_bounding_box())
```

McCrearyD, 2019-06-13:
Good call, I was unaware gdspy automatically changed it's units.

Thank you.

closed 2019-06-13 by McCrearyD