heitzmann/gdspy #91: Duplicate Points Break FlexPath

SBuercklin, 2019-06-21:
If you try to add two consecutive points which are the same to a FlexPath, there's a divide-by-zero problem that leads to `NaN`s in the resulting polygon's vertices.

```
import gdspy 

c = gdspy.Cell("REDUNDANT_PATH")
pts = [[1,1],[1,2],[1,2],[2,2]]
fp = gdspy.FlexPath(pts, .5)

c.add(fp)
polys = c.get_polygons()
print(polys)

gdspy.write_gds("redundant.gds")
```

and output

```
/path/to/packages/gdspy/__init__.py:2835: RuntimeWarning: invalid value encountered in true_divide
  un = un[:, ::-1] * _mpone / ((un[:, 0]**2 + un[:, 1]**2)**0.5).reshape((un.shape[0], 1))
/path/to/packages/gdspy/__init__.py:2861: RuntimeWarning: invalid value encountered in true_divide
  vn = vn[:, ::-1] * _mpone / ((vn[:, 0]**2 + vn[:, 1]**2)**0.5).reshape((vn.shape[0], 1))
[array([[0.75, 1.  ],
       [1.25, 1.  ],
       [1.25, 2.25],
       [ nan,  nan],
       [ nan,  nan],
       [0.75, 1.75],
       [2.  , 1.75],
       [2.  , 2.25],
       [0.75, 2.25],
       [ nan,  nan],
       [ nan,  nan],
       [0.75, 2.25]])]
```

heitzmann, 2019-06-24:
Thanks @SBuercklin, this should be an easy fix. I'll get to it as soon as I have some free time from work.

heitzmann, 2019-06-26:
I believe it should be working now. Please let me know so I can close the issue.

SBuercklin, 2019-06-27:
Seems fixed to me, thanks!

closed 2019-06-27 by SBuercklin