heitzmann/gdspy #92: Using array as a fillet radius

amanagrawal0250, 2019-06-25:
I am trying to use array of 4 elements as a fillet radius so that I can round the four corners of my polygon with four different radius, but I am getting the following error.

**line 448, in fillet
    l = radius * tt[ii]

TypeError: can't multiply sequence by non-int of type 'numpy.float64'**

Please explain how to pass an array as an argument for the fillet.

Currently, my program looks like this:
    po1=[(x-sq/2,y+sq/2-tw/1.414), (x-mem/2,y+mem/2-tw/1.414), (x-mem/2,y-mem/2+tw/1.414), (x-sq/2,y-sq/2+tw/1.414)]; #Just four coordinates for polygon
    poly1=gdspy.Polygon(po1, 2);
    poly1.fillet([100, 100, 25, 25]);

Thanks.

heitzmann, 2019-06-25:
@amanagrawal0250 It looks like you're using a older version of Gdspy. This feature is available only in version 1.4.

amanagrawal0250, 2019-06-25:
@heitzmann Thank you for your response. But, I could not find version 1.4 in Anaconda navigator. I am using Anaconda 1.9.7 with Windows 10.

heitzmann, 2019-06-25:
You can find it here: https://github.com/heitzmann/gdspy/releases

closed 2019-06-29 by heitzmann