heitzmann/gdspy #93: Tkinter error when trying to use Layoutviewer

maelhos, 2019-07-01:
Why when i do :
`if not args.disablepreview:  
    gdspy.LayoutViewer(library=None, cells=[cell])`
python return me a : 
`Traceback (most recent call last):
  File ".\InductorGen.py", line 726, in <module>
    gdspy.LayoutViewer(library=None, cells=[cell])
  File "C:\Users\Maël\Anaconda3\lib\site-packages\gdspy\viewer.py", line 182, in __init__
    self.grid(sticky='nsew')
  File "C:\Users\Maël\Anaconda3\lib\tkinter\__init__.py", line 2226, in grid_configure
    + self._options(cnf, kw))
_tkinter.TclError: cannot use geometry manager grid inside . which already has slaves managed by pack`

heitzmann, 2019-07-02:
@maelhos Are you trying to use `LayoutViewer` inside another Tk application? Can you show more of the code you're using?

maelhos, 2019-07-02:
Here it is [InductorGen](https://github.com/maelhos/InductorGen) just do :
`python .\InductorGen.py -l 200 -t 4 -lt 50 -p 5 -r 0.8 -v 4 -m 1 -d 45 -o tet -s 8`
if you want to try ... and i think this is because of matplotlib ...

maelhos, 2019-07-02:
Ok it works perfectely fine now without matplotlib thank you so much...
But i'm sorry to say that i also need matplotlib, not now but for future graph ...
So only if you can and if you want ... can we implement other tk window compatibilitie ...

heitzmann, 2019-07-02:
TBH I'm not sure what that would require... I'm not used to programming GUI apps. Maybe we can work together on trying to find a solution to your use case.

maelhos, 2019-07-03:
Ok ... Cool, that goes for me.I'm already trying to solve the problem

maelhos, 2019-07-03:
I've asked oubout the tcl error and understood it because of a cool helper on[ stackoverflow](https://stackoverflow.com/questions/56848883/why-gdspy-librarie-give-a-tkinter-tcl-error-when-trying-to-use-the-layout-viewer)... Maybe it could help us ...

heitzmann, 2020-08-02:
No changes after a year, so I'm closing this issue.

closed 2020-08-02 by heitzmann