heitzmann/gdspy #95: Issue while deleting the polygon shapes

lsshruthi, 2019-08-09:
Hi all,

I am new to gdspy and in one of my task I am trying to delete the selected polygons in the cell.   Unfortunately it is not working here. When I give delete polygons with the selected vertices,it is deleting complete polygons in the cell. I am not able to delete the selected shapes based on some condition in the cell. 
 for ex:
cv.remove_polygons(lambda ptlist,layer,datatype: layer == 1))
In the ptlist I am giving only the selected vertices to delete.
ptlist=[[x,y],[a,b]]

Suggestions are appreciated.

Thank you,
Shruthi LS

heitzmann, 2019-08-09:
Hi @lsshruthi,

I believe you misunderstood the argument in `remove_polygons`. It takes an user-defined function that determined whether a given polygon should be removed based on its vertices, layer and datatype. So `ptlist` in your example is the *argument* of your function... you cannot assign a value to it.

Maybe what you want is something more similar to:
```python
def test(pts, layer, datatype):
    """This function must return True only for polygons that should be removed.
       In this case, remove polygons in layer 1 with at least 1 vertex close to
       coordinates (5, 23)"""
    if layer == 1:
        sqdist = ((pts - numpy.array(5, 23))**2).sum(axis=1)
        if sqdist.min() < 1e-8:
            return True
    return False

cv.remove_polygons(test)
```

However, it is not possible to remove vertices from polygons—the whole polygon is removed from the cell. To change the shape of polygons, you should be using other functions, such as `boolean`, `offset`, and `slice`, or, better yet, construct your shapes such that they don't need to be modified latter.

lsshruthi, 2019-08-19:
test(pts, layer, datatype)
shapes =  cv.get_polygons(by_spec = True, depth = 0)
boundary = (shapes[13,0])
obj = (shapes[11,0])
invalid_objs = gdspy.fastboolean(boundary ,obj , 'and' ,layer=63)
#code to delete invalid_objs

I want to delete the invalid_objs of the boolean operation. The remove_polygons function in the init.py mentions "The function test is called for each polygon in the cell " what if we have million shapes to delete will it call that many times which makes execution time slower.

Is this the only way to delete or we have a better option to do.

heitzmann, 2019-08-19:
This is the only function for deleting polygons at this moment... You can always write your own custom function to get better performance in your specific use-case. To delete a polygon from a cell, simply remove it from the list `cell.polygons`.

lsshruthi, 2019-08-20:
Okay. Thank you. Will try doing it.

lsshruthi, 2019-08-27:
layer1 = 1
layer2 = 2
layer3 = 3

objs_to_remove = gdspy.fast_boolean(layer1_polygons,layer2_polygons,"and",layer=3)

# some polygons in layer1 are coincident to those in layer2, which need to be removed.

all_polygons = cellview.get_polygons(depth=1,by_spec=true)
ref_polygon_set = all_polygons[layer1,0]
for obj in objs_to_remove.polygons:
    for polygon in ref_polygons.polygons:
            if (obj[:] == polygon[:]).any(): #does delete the objects of interest, along with any other shape that are co-linear.
            #    if (obj[:] == polygon[:]).all(): # does not find any matches
                   cellview.polygons.remove(item)

In the above code .all does not find any matches. Is the proper way to do. Any error in the above mentioned code..?

lsshruthi, 2019-08-27:
layer1 = 1
layer2 = 2
layer3 = 3
objs_to_remove = gdspy.fast_boolean(layer1_polygons,layer2_polygons,"and",layer = 3)
#some polygons in layer1 are coincident to those in layer2, which need to be removed.
all_polygons = cellview.get_polygons(depth=1,by_spec=true)
ref_polygon_set = all_polygons[layer1,0]
for obj in objs_to_remove.polygons:
     for polygon in ref_polygon_set.polygons:
           if(obj[:]   == polygon[:]).any():   #does delete the objects of interests,along with any other shape that are co-linear.
                  #if(obj[:]    ==  polygon[:]).all():  #does not find any matches.
                           cellview.polygons.remove(item)

In the above code .all does not find any matches. Any error in the code..?

lsshruthi, 2019-08-29:
Okay.Thank you.

closed 2019-08-29 by lsshruthi