heitzmann/gdspy #97: Error: Argument out of range

totyped, 2019-08-13:
I am creating hole arrays, using the code you kindly provided here:  https://github.com/heitzmann/gdspy/issues/96

I would like to create an array of  hole arrays, meaning that I would place a hole array into a cell, and then use a cell array reference in a top_cell: 

`def main():
    ## ------------------
    ## --   Geometry   --
    ## ------------------
    
    # Sample
    diameter = 300
    period = 600
    gap = 50
    
    # sample size
    N = 167
    ## --------------
    ## --Create GDS--
    ## --------------
    
    top_cell = gdspy.Cell('top_cell')
   
    
    # Layer 1
    array_cell = gdspy.Cell('array_cell')
    hole_array_reference = hole_array(pos=(0,0),diameter=diameter,period=period,Nx=N,Ny=N,name = "hole_array",layer_number=0)
    array_cell.add(hole_array_reference)
    
    cell_array_ref = gdspy.CellArray(array_cell, 3, 1, (1e6, 0), (0, 0))
    
    top_cell.add(cell_array_ref)

    
    
    # Save GDS
    ## Specify Unit
    #gdspy.LayoutViewer()
    gdspy.write_gds('array.gds', unit=1.0e-9, precision=1.0e-12)
    

    # View GDS (if small, else open KLayout)
    #gdspy.LayoutViewer()

if __name__ == '__main__':
    main()`

As you can see in the code, I would like to have a 2mm period between those arrays. However, due to the limited range of coordinates that can be stored (see here: https://gdspy.readthedocs.io/en/stable/gettingstarted.html?highlight=precision#gdsii-library), it is not possible.

Of course, I could use a gds editor like Klayout to manually copy and paste the gds array, but Iit would be much more convinient if I could do it with gdspy. Therefore, i would mlike to ask you, if you see any way of doing it in gdspy? 

Thank you.

heitzmann, 2019-08-13:
Hello @totyped. Unfortunately, other editors would no solve your problem, because that's a limitation in the GDSII format itself. The only way to write beyond the 2mm range is to lower your precision from 1.0e-12 to 1.0e-11. To be honest, I find it hard to believe that you need picometer resolution in you layout, whatever fabrication technique you're using, so it should not be a problem in practice.

totyped, 2019-08-13:
I see. Thank you very much for your help!  Yes, actually I will be using e-beam, I can even set the precision to 1nm. I somehow did not think of that... Thanks a lot!

closed 2019-08-13 by totyped