heitzmann/gdspy #102: Filleting with voids error

trezitorul, 2019-10-08:
When filleting a polygon with an enclosed void. The fillet function fillets the intersection between the line connecting the void to the outside of the polygon. 
[image]

heitzmann, 2019-10-08:
@trezitorul This is a known limitation... Is it possible for you to fillet before creating the holes? Otherwise, the solution would be to create small rectangles to patch the undesired fillets (using a boolean union).

trezitorul, 2019-10-09:
Ah I see, so when making complex structures like this it is best to keep everything as independent shapes and at the end integrate and fillet.

heitzmann, 2019-10-10:
Yes, it strongly depends on the layout. Detecting and deciding whether such edges/vertices should be included in the fillet operation is a hard problem and I can see situations where the user would indeed want the operation to be performed.

In cases like yours, a quite general solution is to fillet before the boolean subtraction.

closed 2019-11-27 by heitzmann