heitzmann/gdspy #103: Creating an array of cells

jethiercrc, 2019-10-24:
This is probably a trivial question, but I can't seem to find the answer in the help files.

I have a library of cells (call them cell01, cell02, etc) and I want to make an array of these cells as a final GDS file. 

I have a library of these cells. I then run a for-loop that loops through the (x,y) positions I want these cells to be placed and which cell specifically I want at said location. For each point, I get the polygons of the cell, translate them, and then add to the tiled_cell. In this case, a new copy of every cell is created as opposed to a reference. The resulting GDS file is enormous.

I would much rather place a reference to the cell (e.g. cell01) that also includes the desired translation rather than copy the polygons each time. Any hints on how to accomplish this?

Here's the code:
tiledCell= gdspy.Cell('tiledCell')

 for indx in range(NUM_X):
   for indy in range(NUM_Y):
     Lc = int(designLengths[indx][indy])   
     tile = (ElementLibrary.top_level()[Lc]).get_polygons()
     TiledSurface.add(gdspy.PolygonSet([[[vertice[0]+indx*latticeSize[0], vertice[1]+indy*latticeSize[0]] for vertice in polygon] for polygon in tile]))

heitzmann, 2019-10-24:
Hello, @jethiercrc. You should be using [`CellReference`](https://gdspy.readthedocs.io/en/stable/reference.html#cellreference) (maybe even [`CellArray`](https://gdspy.readthedocs.io/en/stable/reference.html#cellarray), but I'm not sure what `designLenghts` is) to avoid copying the geometry:

```python
tiledCell= gdspy.Cell('tiledCell')
cellLib = ElementLibrary.top_level()

for indx in range(NUM_X):
  for indy in range(NUM_Y):
    Lc = int(designLengths[indx][indy])
    tiledCell.add(gdspy.CellReference(cellLib[Lc], (indx * latticeSize[0], indy * latticeSize[1])))
```

There is a section in the tutorial about references: https://gdspy.readthedocs.io/en/stable/gettingstarted.html#references

jethiercrc, 2019-10-24:
This looks very promising - designlengths is just an index and would reference the cell in the library.

I'll give this a try and report back.

jethiercrc, 2019-10-24:
In terms of what I need to save using write_gds - do I need to save more than [tiledCell] ?
I tried saving [tiledCell] alone and the GDS file has all the refs, but the cells themselves are empty.

heitzmann, 2019-10-24:
Yes, you have to include ass referenced cells and their dependencies in the tiled layout.
The easiest way of doing so is through [`GdsLibrary.extract`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.GdsLibrary.extract):

```python
for c in cellLib:
  ElementLibrary.extract(c)
```

This will change in future versions of Gdspy, though, but for now it is the easiest choice.

jethiercrc, 2019-10-25:
So I have tiledCell to save to the GDS file (the collection of references to the cells) but when I save tiledCell to GDS none of the cell's themselves are saved. If I run extract on the cells, do I need to save something other than [tiledCell] in order to get the cells into the GDS file? (I still don't see the cells)

heitzmann, 2019-10-25:
I didn't understand that you were passing the argument `[tiledCell]` to `write_gds`. You don't need to pass that argument, since all created and extracted cells are internally stored in [`current_library`](https://gdspy.readthedocs.io/en/stable/reference.html#current-library). Simply use `gdspy.write_gds("filename.gds")`.

Otherwise, the more robust way of doing this is by creating yourself a [`GdsLibrary`](https://gdspy.readthedocs.io/en/stable/reference.html#gdslibrary):
```python
cellLib = ElementLibrary.top_level()

outLib = gdspy.GdsLibrary()
tiledCell = gdspy.Cell('tiledCell')
outLib.add(tiledCell)

for indx in range(NUM_X):
  for indy in range(NUM_Y):
    Lc = int(designLengths[indx][indy])
    tiledCell.add(gdspy.CellReference(cellLib[Lc], (indx * latticeSize[0], indy * latticeSize[1])))

outLib.add(tiledCell.get_dependencies(True))

outLib.write_gds("filename.gds")
```

jethiercrc, 2019-10-25:
Absolutely wonderful! Thank you for your comments/help/guidance and work you've put into this library.

closed 2019-10-25 by jethiercrc