heitzmann/gdspy #106: Best method flattening the GDS?

ncFoCo, 2019-11-04:
Hello,

First, thank you for developing this library. I think it is exactly what I need to accomplish a project that I am working on. My issue is not with the code in the libray, but more on the correct methodology to use it for reading existing GDS files and extracting location data.

**Problem**
I am trying to write a program that flattens a cell that exists in all of our GDS files. This particular cell describes an entire reticle field. The goal is to send the flattened data into a database that we can query to determine locations of specific cells and structures.

I am easily able to flatten the cell and all of the child references using a recursive method. But, I am struggling with applying the correct x_reflection and rotation. The main issue is that a child reference may indicate that it it x_reflected and rotated, but how do I accurately propagate this down to all the child references since they may have their own x_reflection and rotation?

**Questions**
I wanted to ask a couple of questions before I dive too deep into this project. 
1. Is there an existing or recommended method to flatten out the data maintaining the reference cell locations and names?
2. I feel like there should be a fairly simple recursive method to accomplish this, but the x_reflection and rotation are eluding my capability at the moment?
3. Am I using this library incorrectly for my given problem statement?

Thanks for any help!

heitzmann, 2019-11-05:
Hello @ncFoCo,

If you want to have your own function for flattening the cell, maybe take a look at the source code to see how it handles rotations and reflection. It's indeed a recursive function that gathers the polygons in each reference and apply is own transformation before passing everything to the caller.

I imagine you cannot use [`Cell.flatten`](https://gdspy.readthedocs.io/en/stable/reference.html#gdspy.Cell.flatten) because of your 1st question. I'm not sure what you want to maintain... Can you give a simple example of the output you'd like this flatten function to have?

ncFoCo, 2019-11-05:
Hi @heitzmann,

Thanks for the quick reply. I will spend more time understanding how gdspy applies x_reflection and rotation for Cells. I am fairly confident that I can achieve my desired output since all of the necessary information is available.  

At the end of the day, I don't care about the cell details such as the polygons, paths, and labels. I just need to know where the origin of each cell name is in the reticle. I hope that makes sense?

---
## Library Details
Here is an example library structure that is common in all of our GDS files. 

### Main Library
```
<Cell DieMain>
<Cell DieExp1>
<Cell SubDieStructure>
<Cell ULCorner>
<Cell URCorner>
<Cell AlignmentMarkN>
<Cell AlignmentMarkN+1>
<Cell CornerDumification>
<Cell Reticle>
<Cell Wafer (Top Level)>
```
The `Reticle` cell is the one that I want to flatten such that I can locate all of the other cells based on the cell names.

### Reticle Cell
The `Reticle` and dependencies are structure like this:  
```
<Cell Reticle>
  <CellRef DieMain>
  ...
  <CellRef DieMain>
  <CellRef DieMainMX (x_reflection of DieMain)>
  ...
  <CellRef DieMainMX (x_reflection of DieMain)>
  <CellRef DieExp1>
  ...
  <CellRef DieExp1>
  <CellRef ULCorner>
  <CellRef URCorner>
  ...

<Cell DieMain>
  <CellRef SubDieStructure>
<Cell DieMainMX>
  <CellRef DieMain (x_reflect)>
<Cell DieExp1>
  <CellRef SubDieStructure>
<Cell ULCorner>
  <CellRef AlignmentMarkN>
  <CellRef CornerDumification>
<Cell URCorner>
  <CellRef AlignmentMarkN+1>
  <CellRef CornerDumification (x_reflect, rot=180)>
```

At the end of my algorithm, I want to build a table that looks like this:
```
CELL_NAME        PHYSICAL_X        PHYSICAL_Y        CELL_SIZE_X        CELL_SIZE_Y
DieMain          0                 0                 900                600
SubDieStructure  60                60                120                120
DieMain          900               0                 900                600
SubDieStructure  960               60                120                120
DieMainMX        1800              0                 900                600
SubDieStructure  1860              540               120                120
DieExp1          2700              0                 1800               600
   ...
```

heitzmann, 2019-11-05:
I see. It makes sense and it should be possible. Keep in mind that when you have rotations different from 180° or any magnification, the size of the reference (in terms of bounding box) will also change.

ncFoCo, 2019-11-06:
Closing this issue for the time being as it does not pertain to the main gdspy code base.  

Let me know if you would like to see my implementation for this feature and I can open a pull request.

closed 2019-11-06 by ncFoCo

ncFoCo, 2019-11-13:
For anyone who may need this in the future, here is my solution that I have tested against our GDS data.

```
"""Generic class that extracts all ReferenceCell names and locations from a GDS Cell."""
import numpy as np
import pandas as pd

class GDSCellExtractor:

    def __init__(self, cell=None):
        self.cell = cell

    def flat_df(self, cell=None, *args, **kwargs):
        """Create a DataFrame from a flattened GDS Cell."""

        rows = self.flatten(cell=cell, *args, **kwargs)
        df = pd.DataFrame(columns=['cell_name'])
        df = df.append(rows, ignore_index=True, sort=False)
        if cell is not None:
            name = cell.name
        elif self.cell is not None:
            name = self.cell.name
        else:
            name = None
        df['cell_name'] = name
        df = df.astype({'cell_name': 'category', 'ref_name': 'category'})
        return df

    def to_csv(self, location, cell=None, mirror=True, rotate=True, offset=True, *args, **kwargs):
        df = self.flat_df(cell=cell, mirror=mirror, rotate=rotate, offset=offset)
        df.to_csv(location, *args, **kwargs)

    def plan_to_csv(self, location, cell=None, *args, **kwargs):
        rows = self._translation_plan(cell=cell)
        df = pd.DataFrame(data=rows)
        df.to_csv(location, *args, **kwargs)

    def to_json(self, location, cell=None, mirror=True, rotate=True, offset=True, *args, **kwargs):
        df = self.flat_df(cell=cell, mirror=mirror, rotate=rotate, offset=offset)
        df.to_json(location, *args, **kwargs)

    def flatten(self, cell=None, *args, **kwargs):
        """Flatten all ReferenceCells contained within a GDS Cell."""

        if cell is not None:
            c = cell
        else:
            c = self.cell
        plans = self._translation_plan(cell=c)
        rows = []
        for plan in plans:
            # Apply all of the collected translations for this ReferenceCell
            cell_props = self._apply_cell_plan(plan, *args, **kwargs)
            # Generate the row properties for this Reference Cell
            ref_name = cell_props.get('ref_name')
            origin = cell_props.get('origin')
            bb = cell_props.get('bounds')
            ox, oy = origin
            l, u = bb
            lx, ly = l
            ux, uy = u
            cell_size_x = ux - lx
            cell_size_y = uy - ly
            row = {
                'ref_name': ref_name,
                'cell_size_x': cell_size_x,
                'cell_size_y': cell_size_y,
                'origin_x': ox,
                'origin_y': oy,
                'lower_x': ox + lx,
[... 99 lines cut ...]
        return None

    def rotate_points(self, points, theta=0):
        """Local method to rotate arrays about their origin."""

        theta_r = np.radians(theta)
        s, c = np.sin(theta_r), np.cos(theta_r)
        T = np.array([[c, -s], [s, c]])
        if points.shape == (2, ):
            rotated = T.dot(points)
        else:
            rotated = [T.dot(point) for point in points]
            rotated = np.array(rotated)
        rotated = np.round(rotated, 3)
        rotated = self.sort_bounds(rotated)
        return rotated

    def mirror_points(self, points):
        """Local method to mirror array about the x-axis."""

        mirrored = points * np.array([1, -1])
        mirrored = self.sort_bounds(mirrored)
        return mirrored

```