heitzmann/gdspy #107: [Bug] After Flatten, FlexPath instances in Rotated/Mirrored Sub-Cells in Wrong Places

qrqiuren, 2019-11-08:
Run the following program with the attached GDS file.

```python
# Run with GDSPY 1.4.2
import gdspy as gds

circuit_gds = gds.GdsLibrary()
circuit_gds.read_gds('RING_OSC.gds')
inv = circuit_gds.cell_dict['TRAINING_INVERTER']
cell = circuit_gds.cell_dict['RING_OSC']
cell_origin = cell.copy('RING_OSC_ORIGIN')
cell.flatten()
gds.LayoutViewer(cells=[inv, cell_origin, cell])
```

Example GDS:
[RING_OSC.zip](https://github.com/heitzmann/gdspy/files/3824294/RING_OSC.zip)

The RING_OSC cell is a ring oscillator with 4 inverters inside, and 12 inverters as dummies.
We can see that after flatten, some paths from rotated/mirrored sub-cells (all in layer 31/0) misplaced, as in the red circles.

[image]

heitzmann, 2019-11-08:
Thanks for the bug report! I'll look into it asap!

heitzmann, 2019-11-11:
Fixed in version 1.4.3.

closed 2019-11-11 by heitzmann